ralphwetzel/node-red-mcu-plugin #1: Error when running plugin

mtoko, 2022-08-14:
Thank you for all the work on this plug in
I have installed as instructed in the standard Node Red directory (~/.node-red) and the plugin can be seen in the node_modules directory in a folder @ralphwetzel.
When starting Node Red the following error appears in the console and the plug in is not displayed in the Node Red Editor

`TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received undefined
    at new NodeError (node:internal/errors:372:5)
    at validateString (node:internal/validators:120:11)
    at Object.join (node:path:1172:7)
    at module.exports (/home/darryl/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/mcu_plugin.js:349:34)
    at loadPlugin (/usr/local/lib/node_modules/node-red/node_modules/@node-red/registry/lib/loader.js:419:27)
    at /usr/local/lib/node_modules/node-red/node_modules/@node-red/registry/lib/loader.js:458:31
    at Array.forEach (<anonymous>)
    at loadNodeSetList (/usr/local/lib/node_modules/node-red/node_modules/@node-red/registry/lib/loader.js:455:11)
    at /usr/local/lib/node_modules/node-red/node_modules/@node-red/registry/lib/loader.js:143:16 {
  code: 'ERR_INVALID_ARG_TYPE'`

ralphwetzel, 2022-08-14:
Thank you for testing this plugin!
Please install the Moddable SDK according to the [Getting Started Guide](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md).
Please ensure especially, that you define the MODDABLE environment variable as described in [this chapter](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#installing).

The error you hit is due to the fact that MODDABLE currently is not defined on your system. Thank you for catching this! I will add a guard to compensate for this situation.

mtoko, 2022-08-14:
Thank you for the advice
I have had the Moddable SDK installed and working with the hello world example and have used the node-red-mcu to run flows on esp 32. The MODDABLE environment variable appears to be correct. When I input cd $MODDABLE the directory is ~/Projects/moddable.
This is the extract from the PATH
/home/darryl/Projects/moddable/build/bin/lin/release:
/home/darryl/Projects/iot-product-dev-book:
/home/darryl/Projects/node-red-mcu:
/home/darryl/Projects/moddable/module

ralphwetzel, 2022-08-14:
@mtoko
Thank you for these insights.
Could you please pull the latest update (w/ the guard in place) & check, if it returns an error?

mtoko, 2022-08-14:
I have pulled the latest update and the error is not returned but a message 
14 Aug 09:21:28 - [error] *** node-red-mcu-plugin:
14 Aug 09:21:28 - [error] Environment variable $MODDABLE is not defined or stating a non-existing path.
14 Aug 09:21:28 - [error] Please install the Moddable SDK according to its Getting Started Guide:
14 Aug 09:21:28 - [error] https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md

I have tested the Moddable installation and can copy a flow from the editor and paste into flows.js and this works. Does this indicate that the Moddable setup is correct?

ralphwetzel, 2022-08-14:
I'm sure your Moddable setup is correct as you've already worked with it successfully.
On the other hand, there might be an issue with the definition of the environment your node-red is running in.

To sort this out, I would like to ask you to add a logging statement @ line 30 of your **installed** `mcu_plugin.js`:

Original code:
https://github.com/ralphwetzel/node-red-mcu-plugin/blob/da304486000dd9961f04ea323afd58c0fb145fdc/mcu_plugin.js#L28-L33

With modification:
```
  // ***** 
  // Ensure that MODDABLE is defined 
  console.log(process.env);
  const MODDABLE = process.env.MODDABLE;
```

After restart of node-red, you'll find the environment variables it knows about printed out to your console (prior to the error message you already know). Does this comprise MODDABLE ?

mtoko, 2022-08-14:
I have added the console.log and this is the output
`{
  LANGUAGE: 'en_GB:en',
  USER: 'darryl',
  HOME: '/home/darryl',
  OLDPWD: '/usr/local/lib/node_modules/node-red/bin',
  NODE_OPTIONS: '--max_old_space_size=2048',
  LOGNAME: 'darryl',
  JOURNAL_STREAM: '8:51772',
  PATH: '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin',
  INVOCATION_ID: '4c7c10959695415e813341f8222bcd8f',
  LANG: 'en_GB.UTF-8',
  SHELL: '/bin/bash',
  PWD: '/home/darryl',
  XDG_DATA_DIRS: '/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/',
  NODE_RED_HOME: '/usr/local/lib/node_modules/node-red'
}
14 Aug 17:38:17 - [error] *** node-red-mcu-plugin:
14 Aug 17:38:17 - [error] Environment variable $MODDABLE is not defined or stating a non-existing path.
14 Aug 17:38:17 - [error] Please install the Moddable SDK according to its Getting Started Guide:
14 Aug 17:38:17 - [error] https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md
`

ralphwetzel, 2022-08-14:
Thank you for providing this data. As I anticipated, this extract shows that your Node-RED doesn't know the $MODDABLE environment variable.
There's a [chapter](https://nodered.org/docs/user-guide/environment-variables#running-as-a-service) in the Node-RED documentation [ that I just got aware of] that matches this situation. In short:

> When Node-RED is running as a service having been installed using the provided [script](https://nodered.org/docs/getting-started/raspberrypi), it will not have access to environment variables that are defined only in the calling process.

The _solution_ is stated as well:

> In this instance, environment variables can be defined in the settings file by adding `process.env.FOO='World';`
placed outside the module.exports section. Alternatively, variables can be defined as part of the systemd service by placing statements of the form `ENV_VAR='foobar'` in a file named environment within the Node-RED user directory, `~/.node-red`.

If you confirm this proposal solves the issue, I'll add a dedicated comment to the error message.

mtoko, 2022-08-15:
I have added the environment variable to the settings.js at the top of the file before the module.exports

process.env.MODDABLE='$MODDABLE',
module.exports = {

/*****************************************************************************>
 * Flow File and User Directory Settings
 *  - flowFile

When starting Node Red the error message is displayed but the console.log indicates MODDABLE

{
  LANGUAGE: 'en_GB:en',
  USER: 'darryl',
  HOME: '/home/darryl',
  OLDPWD: '/usr/local/lib/node_modules/node-red/bin',
  NODE_OPTIONS: '--max_old_space_size=2048',
  LOGNAME: 'darryl',
  JOURNAL_STREAM: '8:103239',
  PATH: '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin',
  INVOCATION_ID: '296c77db59f048ce8628bbde557ea29b',
  LANG: 'en_GB.UTF-8',
  SHELL: '/bin/bash',
  PWD: '/home/darryl',
  XDG_DATA_DIRS: '/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/',
  NODE_RED_HOME: '/usr/local/lib/node_modules/node-red',
  MODDABLE: '$MODDABLE'
}
15 Aug 08:16:25 - [error] *** node-red-mcu-plugin:
15 Aug 08:16:25 - [error] Environment variable $MODDABLE is not defined or stating a non-existing path.
15 Aug 08:16:25 - [error] Please install the Moddable SDK according to its Getting Started Guide:
15 Aug 08:16:25 - [error] https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md

The MCU sidebar is available and a new build configuration can be added. I have used esp32 but the @ box is empty and there is no result when the Build button is selected

ralphwetzel, 2022-08-15:
I'm pretty sure, you have to provide the full path to the moddable directory rather than a reference to another env.variable. That said, I propose you add the following line to your settings.js:

`process.env.MODDABLE = "/home/darryl/Projects/moddable"`

> The MCU sidebar is available [...].

After you hit that error, the code should have bailed out. I'll verify that once again.
Thank you for giving this plugin a try!

mtoko, 2022-08-15:
Thank you for the information, I am making progress. I have added the environment variable as proposed and mcu loads without errors. I have added a build configuration for the usb port that is connected to an esp32 nodemcu. When I attempt the build mcconfig is not found. See the screenshots below

[image]

[image]

ralphwetzel, 2022-08-16:
That's looks like another issue with the env.
Your
`PATH: '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin',`
doesn't include the path to the MODDABLE tools directory, as demand according to the [documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#installing-2) @ Installing, ch 7:
`export PATH=$PATH:$MODDABLE/build/bin/lin/release`.

I've pushed another change that checks now if it is defined & adds it to the runtime's environment if not.

mtoko, 2022-08-16:
Thank you
I have checked the documentation and my ~/.bashrc file and the export PATH=$PATH:$MODDABLE/build/bin/lin/release entry is included.
The new change has been installed and now mcconfig is found, but there is a new error

cd /tmp/node-red-mcu-plugin2IvF05
mcconfig -d -x localhost:5004 -m -p esp32/nodemcu
### Error: '/home/darryl/Projects/moddable/modules/base/structuredClone/manifest.json': manifest not found!
I have then done a git pull of the latest Moddable and the manifest error has gone. The error is now

**cd /tmp/node-red-mcu-pluginxVumbk
mcconfig -d -x localhost:5004 -m -p esp32/nodemcu
fatal: not a git repository (or any of the parent directories): .git
/home/darryl/Projects/moddable/build/tmp/esp32/nodemcu/debug/node-red-mcu-pluginxVumbk/makefile:407: Could not detect ESP-IDF version.
Not enough parameters
Expected ESP IDF v4.4.2, found
make: *** [/home/darryl/Projects/moddable/build/tmp/esp32/nodemcu/debug/node-red-mcu-pluginxVumbk/makefile:825: idfVersionCheck] Erro**r 1

ralphwetzel, 2022-08-16:
> I have checked the documentation and my ~/.bashrc file and the export PATH=$PATH:$MODDABLE/build/bin/lin/release entry is included.

It is ... but Node-RED doesn't have access to it - as explained before.

Your latest error seems to be related to ESP-IDF. There's a [commit](https://github.com/Moddable-OpenSource/moddable/commit/7f3008543c1894df13a68183ad3e195f92a23444) that bumped the required version to 4.4.2. Have you updated this as well?

mtoko, 2022-08-16:
I have checked the ESP-IDF version is 4.4.2 and the error message states 

Expected ESP IDF v4.4.2, found

ralphwetzel, 2022-08-16:
Are you able to create a successful build when targeting a simulated device?

mtoko, 2022-08-16:
[image]

ralphwetzel, 2022-08-16:
OK. Thanks for trying. This comes from node-red-mcu ... yet doesn't help further.

Looking at the Moddable source, it says:
`idfVersionCheck:
	python $(PROJ_DIR_TEMPLATE)\versionCheck.py $(EXPECTED_ESP_IDF) $(IDF_VERSION) || (echo "Expected ESP IDF $(EXPECTED_ESP_IDF), found $(IDF_VERSION)" && exit 1)`

This brings us back to the fact, that it cannot find IDF:
`Expected ESP IDF v4.4.2, found` ... missing the `$(IDF_VERSION)` that should have been found.

I suspect - again - an issue with the PATH definition. Checking PATH on my (Mac) system, there's a bunch of entries related to ESP:

```
    [
      '/Users/ralph/.local/share/esp32/esp-idf/components/esptool_py/esptool',
      '/Users/ralph/.local/share/esp32/esp-idf/components/espcoredump',
      '/Users/ralph/.local/share/esp32/esp-idf/components/partition_table',
      '/Users/ralph/.local/share/esp32/esp-idf/components/app_update',
      '/Users/ralph/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin',
      '/Users/ralph/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin',
      '/Users/ralph/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin',
      '/Users/ralph/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin',
      '/Users/ralph/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin',
      '/Users/ralph/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin',
      '/Users/ralph/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin',
      '/Users/ralph/.espressif/python_env/idf4.4_py3.9_env/bin',
      '/Users/ralph/.local/share/esp32/esp-idf/tools',
      ...
    ]
```
None of this is existing in the PATH info you provided earlier. 
I'll try to find a way to get this into the Node-RED env.

mtoko, 2022-08-16:
Firstly, apologies.

I have added to the settings file and then restarted Node Red. This is the settings file and the bashrc file. If I build with the Simulator I get this error

### cd /tmp/node-red-mcu-pluginyDNo8J
mcconfig -d -x localhost:5004 -m -p sim/moddable_two
# error: Node-RED MCU does not currently run on Linux !
### Error: incompatible platform!

And if I build with ESP32 Node MCU, I get this error. I have checked and idf.py is in /home/darryl/esp32/esp-idf/tools

### cd /tmp/node-red-mcu-pluginSX1pFK
mcconfig -d -x localhost:5004 -m -p esp32/nodemcu
# Reconfiguring ESP-IDF...
/bin/sh: 1: idf.py: not found
make: *** [/home/darryl/Projects/moddable/build/tmp/esp32/nodemcu/debug/node-red-mcu-pluginSX1pFK/makefile:788: /home/darryl/Projects/moddable/build/tmp/esp32/nodemcu/debug/node-red-mcu-pluginSX1pFK/xsProj-esp32/build/config/sdkconfig.h] Error 127

[image]

[image]

ralphwetzel, 2022-08-16:
> Firstly, apologies.

No need to apologize at all. I'm grateful of your support to sort out this issue.
That said, I think I have an idea of what is necessary to get rid of these errors. It yet requires a bit of work & thus a bit of time to integrate the changes. I'll keep you updated...

ralphwetzel, 2022-08-25:
@mtoko I've just pushed some major changes into the repo. This addresses especially your issue that demands additional steps to prepare the environment. Could you please test these latest changes & give me a feedback? Thanks in advance!

mtoko, 2022-08-25:
Thanks, I will test and send feedback

mtoko, 2022-08-28:
I have completed some testing

Prerequisites
I have
1.) Upgraded Moddable to capture any changes since last test
2.) Re-installed node-red-mcu plugin into ~/.node-red to capture any changes
3.) I have run the test on two different Linux machines
4.) I have restarted the machines and Node Red

On the first machine the Node Red Projects feature was enabled and the MCU side bar does not appear.

[image]

On the second machine the Node Red Projects feature is not enabled and the MCU side bar appears. I have tested with two configurations, one with the simulator and one with a ESP32/nodemcu These are the build configurations and the errors

[image]
[image]
[image]
[image]

ralphwetzel, 2022-09-30:
@mtoko -- I was finally able to provide a fix this issue. 

Please be aware that you might need to update your toolchain. The [documentation](https://github.com/ralphwetzel/node-red-mcu-plugin#raspberry-pi-additional-preparation-steps) explains the details.

Looking forward hearing from your progress...

mshioji, 2022-11-26:
Hello, thank you for the great tool!
I'm trying Moddable MCU and your Node-RED Plugin then facing problem close to above issue so I thought I would write on his behalf.

In my case, installed in Ubuntu. (I tried Ubuntu based Linux mint first, then try on original Ubuntu, but results are the same.)
build process stops with: /bin/sh: 1: source: not found

however, it looks build commands are generated properly in the plugin window.
I could build manually from Terminal by copy&paste, and project works on hardware.
 > cd /home/mshioji/.node-red/mcu-plugin-cache/config15lb9d7r1eb
 > source $IDF_PATH/export.sh && mcconfig -d -x localhost:5004 -m -p esp32/moddable_two
 
 in your additional explanation on the document:
1) add MODDABLE as environment variable to your settings.js.
   done.
2) updating your toolchain - to the latest release/v4.4 branch.
   now Moddable guides "git checkout v4.4.2 " at installation guide so I think it's OK.

* I'm not sure but I think it looks "source" command doesn't work properly in the plugin window on Ubuntu.

There is one more thing, I also tried on Mac BigSur and it works fine!
Thank you again for the great tool.
mshioji

ralphwetzel, 2022-11-26:
@mshioji --
Thank you for this report.
Could you please paste here the complete log from the plugin's console window up to the error?! Thank you in advance!

ralphwetzel, 2022-11-26:
> /bin/sh: 1: source: not found

It looks like `/bin/sh` uses `.` as equivalent to `source` (in `/bin/bash`).
The plugin yet should have defined the shell to be used as `/bin/bash`. I've found something that looks like a bug & fixed that. Could you please check if this changes anything for your Ubuntu setup?

mshioji, 2022-11-27:
Hi @ralphwetzel,

Thank you for your quick action, I tried your update and get a lot of information on console window.
it says :
    Checking "python" ...
    Python 3.10.6
    "python" has been detected
    Adding ESP-IDF tools to PATH...
    Using Python interpreter in /usr/bin/python
    ...
    The following Python requirements are not satisfied:
    ...
    IDF_PYTHON_ENV_PATH: (not set)

when I look at /usr/bin/python, there is 3.10.6 installed but I don't care this since I using pyenv for Python environment.
on the other hand, when I open terminal manually,  it says:
    Detecting the Python interpreter
    Checking "python" ...
    Python 3.7.7  (this is the version I'm using with pyenv)
    "python" has been detected

do you think this is the reason?

here is the complete log from the console window:
Starting build process...
Host system check: #59-Ubuntu SMP Mon Oct 17 18:53:30 UTC 2022
HOME directory check: /home/mshioji
Creating build environment for platform esp32/moddable_two.
Working directory: /home/mshioji/.node-red/mcu-plugin-cache/config15lb9d7r1eb
$IDF_PATH identified: /home/mshioji/esp32/esp-idf
> cd /home/mshioji/.node-red/mcu-plugin-cache/config15lb9d7r1eb
> source $IDF_PATH/export.sh && mcconfig -d -x localhost:5004 -m -p esp32/moddable_two
Detecting the Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /usr/bin/python
Checking if Python packages are up to date...
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 0.1.43ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
idf-component-manager~=1.0
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
python-socketio<5
jinja2<3.1  # See https://github.com/espressif/esp-idf/issues/8760
itsdangerous<2.1
kconfiglib==13.7.1
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
construct==2.10.54
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: /usr/bin/python
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /home/mshioji/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/mshioji/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/mshioji/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/mshioji/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/mshioji/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/mshioji/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/mshioji/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/home/mshioji/esp32/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/home/mshioji/Projects/moddable/build/bin/lin/release

mshioji, 2022-11-27:
To make sure "which Python is used" is only an issue, 
I installed shown modules on log file into: Python 3.10.6 in /usr/bin/python. 

after this, build process works fine!!

In fact, it would be better to use the Python environment which installed in ESP-IDF installation process on Moddable,  
but I will use it this way for the time being.

Thanks to your great work and support.
mshioji

ralphwetzel, 2022-11-27:
`source $IDF_PATH/export.sh` simply checks if a small script successfully runs when called with `python` as execution binary. Whatever Python installation answers this call will be taken to run the following scripts. If none answers, it tries to do the same with `python3` (which is nice yet quite useless later - but that's another finding).
Unfortunately this is a feature of the Espressif tool chain, that hardly can be customized or manipulated.

mshioji, 2022-11-28:
Thank you for the analysis. So that's how it works internally.
Perhaps what I did could be a workaround?
In any case, I am very comfortable now!
Thank you again,

ralphwetzel, 2022-12-19:
@mshioji
To update you on this: The plugin [meanwhile](https://github.com/phoddie/node-red-mcu/discussions/46#discussioncomment-4443559) takes a defined  `IDF_PYTHON_ENV_PATH` into consideration.

ralphwetzel, 2022-12-21:
@mtoko @mshioji
I have the impression that things have settled nicely. Can we close this issue?

mshioji, 2022-12-21:
Yes, in my side It's fine you can close this issue.
I have tried on another Ubuntu and it works very nice.
Thank you for your work!!

mshioji, 2022-12-24:
Sorry for posting here again.
I tried a clean install to check the installation procedure again but it looks it doesn't work now.
-- This part seems to be quite difficult.
The logs point to python 3.10.6, which is the same as before, the one I don't normally use. The requirements are met, but now the build will stop. (build manually with plugin generated commands on another terminal, it works.)

-- here's complete log:
Starting build process...
Host system check: #62-Ubuntu SMP Tue Nov 22 19:54:14 UTC 2022
MCU Build system check: p1.0.0-rc.2 + #df00a1e @ m3.5.1-82-g0f37f87
HOME directory check: /home/mshioji
Creating build environment for platform esp32/nodemcu.
Working directory: /home/mshioji/.node-red/mcu-plugin-cache/egepdq21qbc
$IDF_PATH identified: /home/mshioji/esp32/esp-idf
$IDF_PYTHON_ENV_PATH is not defined.
> cd /home/mshioji/.node-red/mcu-plugin-cache/egepdq21qbc
Creating build script file...
> /bin/bash ./build.sh
>> source /home/mshioji/esp32/esp-idf/export.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /usr/bin/python
Checking if Python packages are up to date...
Python requirements from /home/mshioji/esp32/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/mshioji/esp32/esp-idf/components/esptool_py/esptool
  /home/mshioji/esp32/esp-idf/components/espcoredump
  /home/mshioji/esp32/esp-idf/components/partition_table
  /home/mshioji/esp32/esp-idf/components/app_update
  /home/mshioji/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin
  /home/mshioji/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin
  /home/mshioji/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin
  /home/mshioji/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin
  /home/mshioji/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /home/mshioji/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
  /home/mshioji/esp32/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

>> mcconfig -d -x localhost:5004 -m -p esp32/nodemcu
Detecting the Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Python requirements from /home/mshioji/esp32/esp-idf/requirements.txt are satisfied.
/home/mshioji/Projects/moddable/build/tmp/esp32/nodemcu/debug/egepdq21qbc/makefile:641: *** IDF_PYTHON_ENV_PATH not set. Try running: source $IDF_PATH/export.sh .  Stop.

ralphwetzel, 2022-12-24:
Please try to update your `Moddable SDK` as well to a current version.
What's the version of ESP-IDF you're currently on?

mshioji, 2022-12-25:
Yes, I've re-install Moddable SDK as well with current version.
The version of ESP-IDF is v4.4.3, I installed it following the link below.
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#lin-instructions

I re-installed all the relevant software this morning, but the situation is the same.
- install moddable sdk
- install esp-idf v4.4.3
- clone node-red-mcu into Projects folder
- npm install node-red-mcu-plugin

ralphwetzel, 2022-12-25:
Whereas I think it doesn't matter for this issue: You don't need to 

> * clone node-red-mcu into Projects folder

`node-red-mcu-plugin` installs this to the expected position in your directory tree.

ralphwetzel, 2022-12-25:
@mshioji : Thank you for this report!

> ```
> Detecting the Python interpreter
> Checking "python" ...
> Python 3.10.6
> "python" has been detected
> Python requirements from /home/mshioji/esp32/esp-idf/requirements.txt are satisfied. [...]

I have the impression, that the cause for the issue is an attempt implemented in the Moddable SDK to address [this topic](https://github.com/phoddie/node-red-mcu/discussions/46#discussioncomment-4401624) - that went a bit too far: According to the `esp-idf` [source](https://github.com/espressif/esp-idf/blob/cfef24863f1acd4418227a727dbb6e356438a385/tools/idf_tools.py#L1617-L1624), `IDF_PYTHON_ENV_PATH` is set only -> **if** <- Python is running in a virtual environment. Your log is telling us, this is not the case with your installation:

> `Using Python interpreter in /usr/bin/python`

Your installation yet is validated by `export.sh` to be ok.

> ```
> >> source /home/mshioji/esp32/esp-idf/export.sh
> ...
> Done! You can now compile ESP-IDF projects.
> ```

This needs - from my perspective - an action of the Moddable team.
@andycarle: Could you please verify!

mshioji, 2022-12-26:
>Whereas I think it doesn't matter for this issue: You don't need to
> - clone node-red-mcu into Projects folder

That's what I thought at first too, but without this step it showed an error like this.
mshioji@blackberry:~/.node-red$ npm install https://github.com/ralphwetzel/node-red-mcu-plugin
npm WARN tarball tarball data for @moddable-node-red/sensor@file:/home/mshioji/Projects/node-red-mcu/nodes/sensor (null) seems to be corrupted. Trying again.
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/mshioji/Projects/node-red-mcu/nodes/sensor/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/mshioji/Projects/node-red-mcu/nodes/sensor/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

and, Thanks for your analysis of build stop.
It looks It's tough since there's many related things...

ralphwetzel, 2022-12-26:
Guessing here:
Could it be that your `npm` is trying to install (from) a cached version of `node-red-mcu`, that you installed before into `/home/mshioji/Projects/node-red-mcu/` - e.g. to test the sensor node?
If so, you probably need to update this as well.
If not, try to delete the directory & see what happens. I'm pretty convinced, the installation will succeed smoothly.

mshioji, 2022-12-26:
Oops, that was the reason. Yes, I was trying node-red-mcu and sensor node without plugin first on this machine.
it looks "npm cache clean --force" or "npm cache verify --force" doesn't work however installation succeeded after remove and re-install the npm. thank you for your advice.

commit da304486 (2022-08-14): Add $MODDABLE guard

Fixes #1