ralphwetzel/node-red-mcu-plugin #2: SyntaxError: Unexpected token '??=' on manifest.js:286

rei-vilo, 2022-10-16:
When launching Node-RED, the **node-red-mcu-plugin** isn't loaded with the following error

> Welcome to Node-RED
> ===================
> 
> 16 Oct 10:36:27 - [info] Node-RED version: v3.0.2
> 16 Oct 10:36:27 - [info] Node.js  version: v14.20.1
> 16 Oct 10:36:27 - [info] Linux 5.10.0-18-amd64 x64 LE
> 16 Oct 10:36:27 - [info] Loading palette nodes
> /home/reivilo/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/lib/manifest.js:286
>             mt.build ??= {};
>                      ^^^
> 
> SyntaxError: Unexpected token '??='
>     at wrapSafe (internal/modules/cjs/loader.js:1001:16)
>     at Module._compile (internal/modules/cjs/loader.js:1049:27)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
>     at Module.load (internal/modules/cjs/loader.js:950:32)
>     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
>     at Module.require (internal/modules/cjs/loader.js:974:19)
>     at require (internal/modules/cjs/helpers.js:101:18)
>     at Object.<anonymous> (/home/reivilo/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/mcu_plugin.js:13:21)
>     at Module._compile (internal/modules/cjs/loader.js:1085:14)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
> 16 Oct 10:36:28 - [info] Dashboard version 3.2.0 started at /ui
> 16 Oct 10:36:28 - [info] Settings file  : /home/reivilo/.node-red/settings.js
> 16 Oct 10:36:28 - [info] Context store  : 'default' [module=memory]
> 16 Oct 10:36:28 - [info] User directory : /home/reivilo/.node-red
> 16 Oct 10:36:28 - [info] Projects directory: /home/reivilo/.node-red/projects
> 16 Oct 10:36:28 - [info] Server now running at http://127.0.0.1:1880/
> 16 Oct 10:36:28 - [info] Active project : Test_MCU
> 16 Oct 10:36:28 - [info] Flows file     : /home/reivilo/.node-red/projects/Test_MCU/flow.json
> 16 Oct 10:36:28 - [warn] Using unencrypted credentials
> 16 Oct 10:36:28 - [info] Starting flows
> 16 Oct 10:36:28 - [info] Started flows
> ^C16 Oct 10:36:50 - [info] Stopping flows
> 16 Oct 10:36:50 - [info] Stopped flows

Commenting the three lines solves the issue.

https://github.com/ralphwetzel/node-red-mcu-plugin/blob/b52b6adf73577a1c3ffb178b817e21e28199db60/lib/manifest.js#L286-L289

ralphwetzel, 2022-10-16:
Thank you for your report!
It looks like an issue w/ the version of `Node.js`. Support for the _Logical nullish assignment (??=)_ operator was implemented (only quite recently) with `v15.0.0`. As Node-RED recommends `v14.x LTS` this feature is nice yet "too new".
I'll provide a fix.

ralphwetzel, 2022-10-16:
Should be fixed now by f9025cb.

ralphwetzel, 2022-10-16:
[Confirmed](https://github.com/phoddie/node-red-mcu/discussions/26#discussioncomment-3890258) to be fixed.

closed 2022-10-16 by ralphwetzel