ralphwetzel/node-red-mcu-plugin #5: #Home is not defined

sleveel, 2022-11-21:
Hello,
Sorry to trouble you with something that might be trivial...

I am running Node Red on a Windows 11 Home edition.
I have installed with some difficulty the plugin (the script didn't work for me so I had to create the directories and do the install by hand).

It now looks like the plugin is installed but when I run it i get the message:
$HOME is not defined.

Am I missing a setting somewhere?

Thanks in advance for your help.

ralphwetzel, 2022-11-21:
> Sorry to trouble you with something that might be trivial...

No problem. Each issue reported is very valuable to us - as it helps to improve this tool.

> I am running Node Red on a Windows 11 Home edition.
[...]
> Am I missing a setting somewhere?

Not really. You're just the first one (that I am aware of) who tries to run this on a Windows platform. As each platform demands some specific steps for setup, those for Windows need some tuning - as it looks like.

You could do me a great favour if you invest a bit of time & a bit more of patience - and act as test bed for this Windows specific setup. I would signal you when changes are implemented that promise some progress & you try to run this on your system - until it's done. The first step e.g. could be available at the end of this week...

sleveel, 2022-11-22:
Dear Ralph,

That would be my pleasure!
My goal is to use a Moddable two and Nore-Red to control some of the
machines we produce so having your plugin work would be a god sent for me.

Looking forward to the update.

All the best,
Stephane.

ralphwetzel, 2022-11-23:
> That would be my pleasure!

Thank you! Here we go:

I've reflected the [Getting Started instructions to build for ESP32 on Windows](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#installing-1) in the plugin now. Given you've followed those, you thus should be able to launch the build process.

Looking forward to the result of your test...

sleveel, 2022-11-23:
ok, here we go.

Step 1:
I removed the previously created folders

Step 2 from the command prompt:
I go to my node-red folder (the user folder as per instructions) and launch
the install.

[image: image.png]

Results:

[image: image.png]
This is what happened before.
I got around it by creating the node-red-mcu-plugin folder myself, doing a
git pull and then calling the install.

log file her below:

0 verbose cli C:\Program Files\nodejs\node.exe
C:\Users\Stephane\AppData\Roaming\npm\node_modules\npm\bin\npm-cli.js
1 info using ***@***.***
2 info using ***@***.***
3 timing npm:load:whichnode Completed in 2ms
4 timing config:load:defaults Completed in 5ms
5 timing
config:load:file:C:\Users\Stephane\AppData\Roaming\npm\node_modules\npm\npmrc
Completed in 6ms
6 timing config:load:builtin Completed in 6ms
7 timing config:load:cli Completed in 3ms
8 timing config:load:env Completed in 1ms
9 timing config:load:file:C:\Users\Stephane\.node-red\.npmrc Completed in
1ms
10 timing config:load:project Completed in 4ms
11 timing config:load:file:C:\Users\Stephane\.npmrc Completed in 0ms
12 timing config:load:user Completed in 1ms
13 timing config:load:file:C:\Users\Stephane\AppData\Roaming\npm\etc\npmrc
Completed in 1ms
14 timing config:load:global Completed in 1ms
15 timing config:load:setEnvs Completed in 2ms
16 timing config:load Completed in 24ms
17 timing npm:load:configload Completed in 24ms
18 timing npm:load:mkdirpcache Completed in 1ms
19 timing npm:load:mkdirplogs Completed in 0ms
20 verbose title npm install
https://github.com/ralphwetzel/node-red-mcu-plugin
21 verbose argv "install" "
https://github.com/ralphwetzel/node-red-mcu-plugin"
22 timing npm:load:setTitle Completed in 2ms
23 timing config:load:flatten Completed in 6ms
24 timing npm:load:display Completed in 8ms
25 verbose logfile logs-max:10
dir:C:\Users\Stephane\AppData\Local\npm-cache\_logs\2022-11-23T20_52_15_615Z-
26 verbose logfile
C:\Users\Stephane\AppData\Local\npm-cache\_logs\2022-11-23T20_52_15_615Z-debug-0.log
27 timing npm:load:logFile Completed in 9ms
28 timing npm:load:timers Completed in 0ms
29 timing npm:load:configScope Completed in 0ms
30 timing npm:load Completed in 49ms
31 silly logfile start cleaning logs, removing 2 files
32 timing config:load:flatten Completed in 0ms
33 timing arborist:ctor Completed in 2ms
34 silly logfile done cleaning log files
35 timing idealTree:init Completed in 142ms
36 verbose shrinkwrap failed to load node_modules/.package-lock.json out of
date, updated: node_modules
37 timing idealTree:userRequests Completed in 5133ms
38 silly idealTree buildDeps
39 silly fetch manifest @ralphwetzel/node-red-mcu-plugin@
[... 493 lines cut ...]
https://github.com/phoddie/node-red-mcu.git node-red-mcu; fi
269 error npm WARN using --force Recommended protections disabled.
269 error node-red-mcu non atteso.
269 error npm ERR! code 1
269 error npm ERR! path
C:\Users\Stephane\AppData\Local\npm-cache\_cacache\tmp\git-clonedTxmIv
269 error npm ERR! command failed
269 error npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c if cd
node-red-mcu; then git pull; else git clone
https://github.com/phoddie/node-red-mcu.git node-red-mcu; fi
269 error
269 error npm ERR! A complete log of this run can be found in:
269 error npm ERR!
C:\Users\Stephane\AppData\Local\npm-cache\_logs\2022-11-23T20_52_33_116Z-debug-0.log
270 verbose exit 1
271 timing npm Completed in 19740ms
272 verbose unfinished npm timer reify 1669236735718
273 verbose unfinished npm timer reify:unpack 1669236749956
274 verbose unfinished npm timer
275 verbose code 1
276 error A complete log of this run can be found in:
276 error
C:\Users\Stephane\AppData\Local\npm-cache\_logs\2022-11-23T20_52_15_615Z-debug-0.log

Il giorno mer 23 nov 2022 alle ore 08:05 ralphwetzel <

ralphwetzel, 2022-11-23:
Your screenshots were not posted.
I propose you file your comments directly in GitHub, rather than sending them via mail.
As a first impression: You were not able to install `node-red-mcu-plugin` on your system - following the [procedure given](https://github.com/ralphwetzel/node-red-mcu-plugin#installation). Is my assumption correct?

sleveel, 2022-11-26:
Yes.
Runing "npm install https://github.com/ralphwetzel/node-red-mcu-plugin" from the comand prompt window, in the .node-red folder didn't work.

I had to create the folder "node-red-mcu" and "node-red-mcu-plugin" and run a "git pull" for each.

It looks like the plugin install correctly.

[image]

However, when I run it I get:

Starting Build process...
Creating build environment for platform sim/moddable_two.
Working directory: \Users\Stephane\.node-red\mcu-plugin-cache\configglzdwp8sucf
$HOME is not defined.

ralphwetzel, 2022-11-26:
> Runing "npm install https://github.com/ralphwetzel/node-red-mcu-plugin" from the comand prompt window, in the .node-red folder didn't work.

That was a very valuable hint. I've changed the install script for `node-red-mcu` to succeed now on any platform supported by node.

sleveel, 2022-12-03:
Hi, I re-installed using the last script. It looks like all is nominal.
Now I need to figure out how to use it...

Thanks!

closed 2022-12-03 by sleveel