ralphwetzel/node-red-mcu-plugin #6: Running the plugin

sleveel, 2022-12-03:
Hi, Now that the Windows 10 install works, I would like to understand the process.

As an example, I setup a flow that displays a timestamp in the dashboard.

I then select this flow, create a configuration for the Simulator of the Moddable Two and run the Build.
The output in the console window looks nominal:

Starting build process...
Host system check: Windows 10 Home
HOME directory check: C:\Users\Stephane
Creating build environment for platform sim/moddable_two.

[image]

This creates a folder and a cache.js file in the .node-red\mcu-pligin-cache folder.

What is the next step from there?

Thanks for the help...
Stephane.

ralphwetzel, 2022-12-03:
It should at least emit the message, that the simulator target is not supported on Windows platforms. Sorry!
You need to connect a physical MCU device to your system & run this process. I'm not sure yet it will succeed. To ensure this is among the prio topics on my current to do list...

ralphwetzel, 2022-12-19:
@sleveel : Windows support for hardware devices is meanwhile fully operational.
I'm looking forward to your feedback...