ralphwetzel/node-red-mcu-plugin #8: Error loading node-red-mcu-plugin

tve, 2022-12-13:
Linux Arch x64, running node-red in docker, only installing node-red-mcu-plugin:
```
Welcome to Node-RED
===================

12 Dec 21:28:15 - [info] Node-RED version: v3.0.2
12 Dec 21:28:15 - [info] Node.js  version: v18.7.0
12 Dec 21:28:15 - [info] Linux 5.15.76-1-MANJARO x64 LE
12 Dec 21:28:16 - [info] Loading palette nodes
*** Error while loading node-red-mcu-plugin:
Failed to calculate correct patch path.
Please raise an issue @ our GitHub repository, stating the following information:
> require.main.path: /usr/src/node-red/node_modules/node-red
> utils.js: /usr/src/node-red/node_modules/node-red/node_modules/@node-red/registry
TypeError [ERR_INVALID_ARG_TYPE]: The "id" argument must be of type string. Received
 null
    at new NodeError (node:internal/errors:387:5)
    at validateString (node:internal/validators:116:11)
    at Module.require (node:internal/modules/cjs/loader:1015:3)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/usr/src/node-red/node-red-mcu-plugin/mcu_plugin.js:129:2
2)
    at Module._compile (node:internal/modules/cjs/loader:1120:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1174:10)
    at Module.load (node:internal/modules/cjs/loader:998:32)
    at Module._load (node:internal/modules/cjs/loader:839:12)
    at Module.require (node:internal/modules/cjs/loader:1022:19) {
  code: 'ERR_INVALID_ARG_TYPE'
}
```

The script to launch the container is below. I have the modaable SDK installed and node-red-mcu-plugin checked out and I map both of these into the docker container.
```
#! /usr/bin/env bash
MOD=/home/sw/moddable
cd $MOD
CMD="set -x; npm i ./node-red-* --omit=dev --no-fund --no-audit"
CMD="$CMD; npm start --cache /data/.npm -- -v --userDir /data"
exec docker run --rm -ti -p 1770:1880 \
  -v $PWD/nr-data:/data \
  -v $PWD/node-red-mcu-plugin:/usr/src/node-red/node-red-mcu-plugin \
  -v $PWD/moddable:/usr/src/moddable \
  -e TZ=America/Los_Angeles \
  -e MODDABLE=/usr/src/moddable \
  --entrypoint bash \
  --name node-red-mcu \
  nodered/node-red:3.0.2-18 \
  -c "$CMD"
```

tve, 2022-12-13:
Looking in the container:
```
bash-5.1$ pwd
/usr/src/node-red
bash-5.1$ find . -name registry
./node_modules/@node-red/registry
```
I ended up patching get_require_path to simply return:
```
    let rm = require.main.path;
    rm = rm.replace(/node_modules.*/, '')
    return rm + req_path
```
This got me through the require issue and now NR starts fine.

ralphwetzel, 2022-12-14:
A few comments - without being sure they're truly relevant:

>   -v $PWD/node-red-mcu-plugin:/usr/src/node-red/node-red-mcu-plugin \

This looks like `node-red-mcu-plugin` is not installed (or mapped) into the `<userdir>/node_modules` folder - despite it should be.

> bash-5.1$ pwd
/usr/src/node-red
bash-5.1$ find . -name registry
./node_modules/@node-red/registry

The location of the `registry` folder as well looks "non-standard". How did you install `node-red` ... and how did you determine the path to map it to for the container?

> require.main.path: /usr/src/node-red/node_modules/node-red

Your system expects to exists under the shown `require.main.path` - whereas it looks like it is mapped to `/usr/src/node-red`.

tve, 2022-12-14:
I'm using the official docker images. Node-RED is installed in /usr/src/node-red and that's ~node-red ($HOME). The user dir is /data. Node-RED has two node_modules directories: the one where it itself is installed (/usr/src/node-red/node_modules) and in the user dir (/data/node_modules). For a npm install it's common to use the home dir. If you install something within the node-red flow editor using the palette manager then it goes into the user dir's node_modules.

The installation of Node-RED is basically:
- cd /usr/src/node-red
- plop down package.json from https://github.com/node-red/node-red-docker/blob/master/docker-custom/package.json
- npm install

Maybe what causes the problem is the way node-red is started:
```
node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS
```
See the package.json file's start command.
Just in case it's not clear, node-red-mcu-plugin ends up in /usr/src/node-red/node_modules/@ralphwetzel/node-red-mcu-plugin

ralphwetzel, 2022-12-15:
Thank you for this data.
I'll update the plugin so that this standard container should be recognized & supported by default.
This yet [doesn't solve the issue](https://github.com/ralphwetzel/node-red-mcu-plugin/discussions/10) w/ xsbug & the sim.

ralphwetzel, 2024-12-29:
Fixed by 0c22628f342b95c24dd3aca82a37ba07a7909e54

closed 2024-12-29 by ralphwetzel