ralphwetzel/node-red-mcu-plugin #9: Need instructions on how to build moddable SDK...

tve, 2022-12-13:
I believe I followed the instructions to install the moddable SDK correctly, but node-red-mcu-plugin doesn't like it... Or perhaps it's node-red-mcu's fault :-)
I'm hitting deploy, trying to build for the simulator and I get:
```
/usr/src/moddable/build/bin/lin/release/mcconfig: line 3: /usr/src/moddable/build/bin/lin/release/tools: No such file or directory
```
Contrary to the error messages, `/usr/src/moddable/build/bin/lin/release/tools` actually exists, but is an empty directory. All the executables are in `/usr/src/moddable/build/bin/lin/debug`. What did I do wrong or what did I omit?

tve, 2022-12-14:
I think the install step didn't complete properly. I ran it without sudo to see where it's trying to put stuff. I'm not thrilled about source packages plastering stuff all over the system, it makes for nightmares down the road. And then when I ran it with sudo I had the feeling it skipped over stuff...

ralphwetzel, 2022-12-16:
Have you been able to clarify this issue?