ralphwetzel/node-red-mcu-plugin #11: Problem building from node-red

Sineos, 2022-12-16:
Just digging into this and trying to get the environment setup. Seems to work basically:
* Can compile hello world from moddable
* Can compile hello world for esp32

Compiling from within node-red does not work:
```
Starting build process...
Host system check: Windows 10 Pro
MCU Build system check: p1.0.0-rc.1 + #89ab684 @ m3.5.1-31-g1e836c7
HOME directory check: C:\Users\souls
Creating build environment for platform esp32/esp32c3.
Working directory: \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
$IDF_PATH is defined: C:/Espressif/frameworks/esp-idf-v4.4.3
$IDF_TOOLS_PATH is defined: C:\Espressif
> cd \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
Creating build batch file...
> cmd.exe build.bat

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat" 
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'
Syntaxfehler.
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3

Adding ESP-IDF tools to PATH...
No directories added to PATH:

C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\;C:\Espressif\tools\idf-git\2.34.2\cmd\;C:\Espressif;C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.34.31933\bin\HostX64\x64;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\VC\VCPackages;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Current\bin\Roslyn;C:\Program Files\Microsoft Visual Studio\2022\Community\Team Tools\Performance Tools\x64;C:\Program Files\Microsoft Visual Studio\2022\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\\x64;C:\Program Files (x86)\Windows Kits\10\bin\\x64;C:\Program Files\Microsoft Visual Studio\2022\Community\\MSBuild\Current\Bin\amd64;C:\Windows\Microsoft.NET\Framework64\v4.0.30319;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\Tools\;C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.23.1\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\frameworks\esp-idf-v4.4.3\tools;C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\;C:\Espressif\tools\idf-git\2.34.2\cmd\;C:\Espressif;C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.34.31933\bin\HostX86\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\VC\VCPackages;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Current\bin\Roslyn;C:\Program Files\Microsoft Visual Studio\2022\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\\x86;C:\Program Files (x86)\Windows Kits\10\bin\\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\\MSBuild\Current\Bin\amd64;C:\Windows\Microsoft.NET\Framework\v4.0.30319;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\Tools\;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Users\souls\AppData\Local\Programs\Python\Python310\Scripts\;C:\Users\souls\AppData\Local\Programs\Python\Python310\;C:\Users\souls\AppData\Local\Microsoft\WindowsApps;C:\Users\souls\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\souls\AppData\Roaming\npm;C:\Users\souls\AppData\Local\GitHubDesktop\bin;C:\Users\souls\Projects\moddable\build\bin\win\release;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\VC\Linux\bin\ConnectionManagerExe;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\VC\Linux\bin\ConnectionManagerExe

Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.3\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Ung�ltige Option - "Users".
Ung�ltige Option - "Users".
Syntaxfehler.
NMAKE : fatal error U1077: 'copy' : return code '0x1'
Stop.

```

What I noticed is this line `[vcvarsall.bat] Environment initialized for: 'x64'`

So I went to `\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n` and edited the `build.bat` file

```
CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"
```

to

```
CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat"

```

Now when starting the `build.bat` in `ESP-IDF 4.4 CMD`, the build seems to work but eventually errors out with another error, which is likely related to the display configuration:

[... 341 lines cut ...]
# xsid commodettoBitmap.h.xsi
# xsid commodettoPoco.h.xsi
# xsid commodettoPocoBlit.h.xsi
# xsid piuAll.h.xsi
# xsid piuMC.h.xsi
# xsid commodettoPocoOutline.h.xsi
# xsl modules
Error: no screen configured
 at setup/piu ()
### Error: no screen configured!
NMAKE : fatal error U1077: 'C:\Users\souls\Projects\moddable\build\bin\win\release\xsl.EXE' : return code '0xd'
Stop.
```

This is my setting in node-red:

[image]

(still waiting for my esp to arrive but I thought some leg work before wouldn't hurt)

Any pointer how to overcome
1. The build issue within node-red
2. How to correct the display setting

would be highly appreciated. Many thanks.

ralphwetzel, 2022-12-16:
Hi @Sineos !
Thank you for testing `node-red-mcu` & this plugin.
Could you please post `C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n\build.bat` here? There shouldn't be a SyntaxError in this file!
As for a generic `esp32c3` no display is configured, you don't need to (and should not) select the `UI Support` option. My proposal is to begin with building a standard (non ui) flow - and then to proceed to examples more advanced.

Sineos, 2022-12-17:
Thanks for coming back to this @ralphwetzel 
Ok, then lets start very basic and step by step:

### Flow:
[image]
```
[
    {
        "id": "238841299fb0cdc1",
        "type": "inject",
        "z": "f5fbb3d44904cdb1",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "5",
        "crontab": "",
        "once": true,
        "onceDelay": "10",
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": true
        },
        "x": 330,
        "y": 220,
        "wires": [
            [
                "ac95e1fcef8cf5a9"
            ]
        ]
    },
    {
        "id": "ac95e1fcef8cf5a9",
        "type": "function",
        "z": "f5fbb3d44904cdb1",
        "name": "function 1",
        "func": "msg.payload = new Date(msg.payload).toISOString()\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "_mcu": {
            "mcu": true
        },
        "x": 480,
        "y": 220,
        "wires": [
            [
                "3cb4eb713e2419aa"
            ]
        ]
    },
    {
        "id": "3cb4eb713e2419aa",
        "type": "mqtt out",
        "z": "f5fbb3d44904cdb1",
        "name": "",
        "topic": "nodemcu/test",
        "qos": "",
        "retain": "",
        "respTopic": "",
[... 526 lines cut ...]
# cc builtinCommon.c.o
# cc digitalbank.c.o
# cc _i2c.c.o
# cc spi.c.o
# cc moddnsparser.c.o
# cc modonewire.c.o
# cc owb.c.o
# cc owb_gpio.c.o
# cc owb_rmt.c.o
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c: In function '_init':
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:388:26: error: incompatible types when assigning to type 'volatile union <anonymous>' from type 'int'
         GPIO.enable_w1ts = (0x1 << gpio_num);
                          ^
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:392:14: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
         GPIO.enable1_w1ts.data = (0x1 << (gpio_num - 32));
              ^~~~~~~~~~~~
              enable_w1ts
NMAKE : fatal error U1077: 'C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.EXE' : return code '0x1'
Stop.

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>
```

C:\Users\souls\Projects\moddable\build\tmp\esp32\esp32c3\release\b33dc2xt18n:
[b33dc2xt18n.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10251124/b33dc2xt18n.zip)

ralphwetzel, 2022-12-17:
I've just pushed a small patch that should detect if your tools are built on `x86` or `x64`.
You should update as well to the latest Moddable SDK version (patched today).
You'll need this to connect to your MCU on Windows.
That done and a MCU at hand, you should be ready to test - successfully!

Sineos, 2022-12-18:
Thanks, but unfortunately it does not work:

### Node-Red
```
Welcome to Node-RED
===================

18 Dec 08:55:24 - [info] Node-RED version: v3.0.2
18 Dec 08:55:24 - [info] Node.js  version: v16.17.0
18 Dec 08:55:24 - [info] Windows_NT 10.0.19045 x64 LE
18 Dec 08:55:24 - [info] Loading palette nodes
18 Dec 08:55:25 - [info] Node-RED MCU Edition Runtime Version: #8ca3f6c
18 Dec 08:55:25 - [info] Node-RED MCU Edition Plugin  Version: v1.0.0-rc.1
18 Dec 08:55:26 - [info] Moddable SDK Version: v3.5.1-57-g7be4b23 (x86)
```

### MCU Build 
```
Starting build process...
Host system check: Windows 10 Pro
MCU Build system check: p1.0.0-rc.1 + #8ca3f6c @ m3.5.1-57-g7be4b23 (x86)
HOME directory check: C:\Users\souls
Creating build environment for platform esp32/esp32c3.
Working directory: \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
$IDF_PATH is defined: C:/Espressif/frameworks/esp-idf-v4.4.3
$IDF_TOOLS_PATH is defined: C:\Espressif
> cd \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
Creating build batch file...
> cmd.exe build.bat

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat" 
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Syntaxfehler.
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3

Adding ESP-IDF tools to PATH...
No directories added to PATH:

C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\;C:\Espressif\tools\idf-git\2.34.2\cmd\;C:\Espressif;C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.34.31933\bin\HostX86\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\VC\VCPackages;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Current\bin\Roslyn;C:\Program Files\Microsoft Visual Studio\2022\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\\x86;C:\Program Files (x86)\Windows Kits\10\bin\\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\\MSBuild\Current\Bin\amd64;C:\Windows\Microsoft.NET\Framework\v4.0.30319;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\Tools\;C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.23.1\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\frameworks\esp-idf-v4.4.3\tools;C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\;C:\Espressif\tools\idf-git\2.34.2\cmd\;C:\Espressif;C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.34.31933\bin\HostX86\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\VC\VCPackages;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Current\bin\Roslyn;C:\Program Files\Microsoft Visual Studio\2022\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\\x86;C:\Program Files (x86)\Windows Kits\10\bin\\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\\MSBuild\Current\Bin\amd64;C:\Windows\Microsoft.NET\Framework\v4.0.30319;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\Tools\;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Users\souls\AppData\Local\Programs\Python\Python310\Scripts\;C:\Users\souls\AppData\Local\Programs\Python\Python310\;C:\Users\souls\AppData\Local\Microsoft\WindowsApps;C:\Users\souls\AppData\ [... cut ...]
NMAKE : fatal error U1077: 'C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.EXE' : return code '0x1'
Stop.

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>
```

`C:\Users\souls\Projects\moddable\build\tmp\esp32\esp32c3\release\b33dc2xt18n`:
[b33dc2xt18n.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10253403/b33dc2xt18n.zip)

`C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n`:
[mcu-plugin-cache.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10253407/mcu-plugin-cache.zip)

ralphwetzel, 2022-12-18:
Could it be that your IDF_PATH definition uses a wrong slash type notation ...

> `$IDF_PATH is defined: C:/Espressif/frameworks/esp-idf-v4.4.3`

... that might be ok for `node.js`, yet `cmd.exe` doesn't like afterwards?

Sineos, 2022-12-18:
> Could it be that your IDF_PATH definition uses a wrong slash type notation ...

I'm not sure where this is coming from:
```
C:\Users\souls>echo %IDF_PATH%
C:\Espressif\frameworks\esp-idf-v4.4.3

C:\Users\souls>
```

also `\\` like
[image]
does not change anything.

And under `ESP-IDF 4.4 CMD` the build starts but errors out finally with 

```
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c: In function '_init':
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:388:26: error: incompatible types when assigning to type 'volatile union <anonymous>' from type 'int'
         GPIO.enable_w1ts = (0x1 << gpio_num);
                          ^
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:392:14: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
         GPIO.enable1_w1ts.data = (0x1 << (gpio_num - 32));
              ^~~~~~~~~~~~
              enable_w1ts
NMAKE : fatal error U1077: 'C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.EXE' : return code '0x1'
Stop.
```

ralphwetzel, 2022-12-18:
I did some further testing using the wrong slash type  & can confirm now that this triggers the `Syntaxfehler`. A small fix is meanwhile in the repo that _should_ address this.

Plus: Could you please select `esp32/moddable_two` as build target?
I can confirm this builds successfully. That done, we'll care then for other targets...

Sineos, 2022-12-18:
Sorry, no luck with either:

### MCU Build
```

Starting build process...
Host system check: Windows 10 Pro
MCU Build system check: p1.0.0-rc.1 + #8ca3f6c @ m3.5.1-57-g7be4b23 (x86)
HOME directory check: C:\Users\souls
Creating build environment for platform esp32/moddable_two.
Working directory: \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
$IDF_PATH is defined: C:\Espressif\frameworks\esp-idf-v4.4.3
$IDF_TOOLS_PATH is defined: C:\Espressif
> cd \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
Creating build batch file...
> cmd.exe build.bat

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat" 
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.24.202208181.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.3\tools
    
Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.3\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Ung�ltige Option - "Users".
Ung�ltige Option - "Users".
Syntaxfehler.
NMAKE : fatal error U1077: 'copy' : return code '0x1'
Stop.
```

### ESP-IDF 4.4 CMD
```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.4.3
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\frameworks\esp-idf-v4.4.3\
[... 475 lines cut ...]
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Espressif/frameworks/esp-idf-v4.4.3/components/mbedtls/port/sha/parallel_engine/esp_sha256.c.obj
ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -IC:/Users/souls/Projects/moddable/build/tmp/esp32/moddable_two/release/b33dc2xt18n/xsProj-esp32/build/config -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/mbedtls/port/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/mbedtls/mbedtls/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/mbedtls/mbedtls/library -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/include/esp_additions/freertos -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/port/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/include/esp_additions -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_hw_support/port/esp32/private_include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/heap/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/log/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/lwip/port/esp32/include/arch -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/soc/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/hal/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_rom/include -IC:/Espressif/frameworks/ [... cut ...]
[180/1035] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/ssl_tls.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
NMAKE : fatal error U1077: 'cd' : return code '0x2'
Stop.
```

`C:\Users\souls\Projects\moddable\build\tmp\esp32\moddable_two\release\b33dc2xt18n`:
[b33dc2xt18n.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10254023/b33dc2xt18n.zip)

`C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n`:
[mcu-plugin-cache.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10254026/mcu-plugin-cache.zip)

ralphwetzel, 2022-12-18:
We'll find this issue... 😀 !

https://github.com/ralphwetzel/node-red-mcu-plugin/blob/aa1665dc07603f5a42151d4422a2389be2c27741/mcu_plugin.js#L1652-L1654

Could you please edit `mcu_plugin.js` and insert @ line 1654 the following code:

`                publish_stdout(JSON.stringify(runner_options) + "\n");
`

... just before `run_cmd`, restart node-red, lauch a build & post the beginning of the log here (up to the start message of the VS Command Prompt. Thank you in advance!

Sineos, 2022-12-18:
Here you go:

```
Starting build process...
Host system check: Windows 10 Pro
MCU Build system check: p1.0.0-rc.1 + #8ca3f6c @ m3.5.1-57-g7be4b23 (x86)
HOME directory check: C:\Users\souls
Creating build environment for platform esp32/moddable_two.
Working directory: \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
$IDF_PATH is defined: C:\Espressif\frameworks\esp-idf-v4.4.3
$IDF_TOOLS_PATH is defined: C:\Espressif
> cd \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
Creating build batch file...
{"cwd":"\\Users\\souls\\.node-red\\mcu-plugin-cache\\b33dc2xt18n","env":{"HOME":"C:\\Users\\souls","PATH":"C:\\Program Files (x86)\\VMware\\VMware Workstation\\bin\\;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\NVIDIA NvDLISR;C:\\Program Files\\Microsoft SQL Server\\Client SDK\\ODBC\\170\\Tools\\Binn\\;C:\\Program Files (x86)\\Microsoft SQL Server\\150\\Tools\\Binn\\;C:\\Program Files\\Microsoft SQL Server\\150\\Tools\\Binn\\;C:\\Program Files\\Microsoft SQL Server\\150\\DTS\\Binn\\;C:\\Program Files (x86)\\Windows Kits\\8.1\\Windows Performance Toolkit\\;C:\\Program Files\\nodejs\\;C:\\Program Files\\Git\\cmd;C:\\Users\\souls\\AppData\\Local\\Programs\\Python\\Python310\\Scripts\\;C:\\Users\\souls\\AppData\\Local\\Programs\\Python\\Python310\\;C:\\Users\\souls\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\souls\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Users\\souls\\AppData\\Roaming\\npm;C:\\Users\\souls\\AppData\\Local\\GitHubDesktop\\bin;C:\\Users\\souls\\Projects\\moddable\\build\\bin\\win\\release;","MODDABLE":"C:/Users/souls/Projects/moddable","BUILD":"C:\\Users\\souls\\Projects\\moddable\\build","PLATFORM":"esp32","SUBPLATFORM":"moddable_two","IDF_PATH":"C:\\Espressif\\frameworks\\esp-idf-v4.4.3","IDF_TOOLS_PATH":"C:\\Espressif","PROCESSOR_ARCHITECTURE":"AMD64"},"signal":{},"windowsHide":true}
> cmd.exe build.bat

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat" 
**********************************************************************
```

and

```
C:\Users\souls>echo %MODDABLE%
C:\Users\souls\Projects\moddable

C:\Users\souls>
```

ralphwetzel, 2022-12-18:
There it is ... in `PATH`:

[image]

Same topic again, wrong slash type.

Sineos, 2022-12-19:
Indeed it works now 👍 
Many thanks for your dedicated support.

What would be the best way to continue with the targets?

ralphwetzel, 2022-12-19:
> Indeed it works now

Top! 👍 

> What would be the best way to continue with the targets?

My proposal: Choose the one that you have in hardware at hand & then try to build a simple flow.
Don't hesitate to call back, if you get into trouble...

Sineos, 2022-12-19:
Currently I would like to build for an esp32c3. As shown above the build fails with 

```
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c: In function '_init':
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:388:26: error: incompatible types when assigning to type 'volatile union <anonymous>' from type 'int'
         GPIO.enable_w1ts = (0x1 << gpio_num);
                          ^
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:392:14: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
         GPIO.enable1_w1ts.data = (0x1 << (gpio_num - 32));
              ^~~~~~~~~~~~
              enable_w1ts
NMAKE : fatal error U1077: 'C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.EXE' : return code '0x1'
Stop.
```

`C:\Users\souls\Projects\moddable\build\tmp\esp32\esp32c3\release\b33dc2xt18n`:
[b33dc2xt18n.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10261982/b33dc2xt18n.zip)

I'm just posting above information "just in case". What would you need to analyze?

ralphwetzel, 2022-12-19:
This looks like an issue (I get the same error when building on my system.) in the [Moddable SDK](https://github.com/Moddable-OpenSource/moddable). You should raise an issue there.
As this one here seems to be resolved, I propose to close it...

Sineos, 2022-12-19:
https://github.com/Moddable-OpenSource/moddable/issues/996

Thanks a lot for your support. Appreciated.

closed 2022-12-19 by Sineos