ralphwetzel/node-red-mcu-plugin #13: Compiling for Moddable Two under Windows

sleveel, 2023-01-29:
Hi there,
I have recently recieved a Moddable Two.
I have loaded the standard examples as per the Moddalbe instructions and everything works.

I am now trying to load a simple Node Red Flow so far without success.
I am wandering if the reason might be that the ESP-IDF PATHs are not setup.
When loading a program following the Moddalbe method, I start by launching a CMD script called ESP IDF 4.4 CMD that looks like this:

Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3

Adding ESP-IDF tools to PATH...
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.3\tools

Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.3\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Espressif\frameworks\esp-idf-v4.4.3>

One then launches the x86 using:
"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat"

Then, one can compile and export to the board using:
mcconfig -d -m -p esp32/moddable_two

Here is the output from the console monitor in Node Red:

Starting build process...
Host system check: Windows 10 Home
HOME directory check: C:\Users\Stephane
Creating build environment for platform esp32/moddable_two.
Working directory: \Users\Stephane\.node-red\mcu-plugin-cache\configytvokt5ta5i
$IDF_PATH is defined: C:\Users\Stephane\esp32\esp-idf
> cd \Users\Stephane\.node-red\mcu-plugin-cache\configytvokt5ta5i
Creating build batch file...
> mcconfig -d -x localhost:5004 -m -p esp32/moddable_two
C:\Users\Stephane\.node-red\mcu-plugin-cache\configytvokt5ta5i>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'
IDF_TOOLS_PATH not set. Setting to C:\Espressif\
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Users\Stephane\esp32\esp-idf\
"null" non � riconosciuto come comando interno o esterno,
 un programma eseguibile o un file batch.
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Users\Stephane\esp32\esp-idf
Adding ESP-IDF tools to PATH...
ERROR: Platform Windows- appears to be unsupported
1 file copiati.
1 file copiati.
1 file copiati.
Reconfiguring ESP-IDF...
Traceback (most recent call last):
  File "C:\Users\Stephane\esp32\esp-idf\tools\idf.py", line 36, in <module>
from idf_py_actions.tools import executable_exists, idf_version, merge_action_lists, realpath  # noqa: E402
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Stephane\esp32\esp-idf\tools\idf_py_actions\tools.py", line 7, in <module>
import click
ModuleNotFoundError: No module named 'click'
NMAKE :
fatal error U1077: 'python' : codice restituito '0x1'
Stop.

ralphwetzel, 2023-01-29:
It looks like your IDF_PATH definition is not consistent:

> Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3

> $IDF_PATH is defined: C:\Users\Stephane\esp32\esp-idf

The second environment doesn't seem to be configured as it should be...

sleveel, 2023-02-04:
Hi, I corrected the PATH.
It looks like I have another stumbling block.
Here is the console output:

Starting build process...
Host system check: Windows 10 Home
HOME directory check: C:\Users\Stephane
Creating build environment for platform esp32/moddable_two.
Working directory: \Users\Stephane\.node-red\mcu-plugin-cache\configytvokt5ta5i
$IDF_PATH is defined: C:\Espressif\frameworks\esp-idf-v4.4.3
> cd \Users\Stephane\.node-red\mcu-plugin-cache\configytvokt5ta5i
Creating build batch file...
> mcconfig -d -x localhost:5004 -m -p esp32/moddable_two ssid="Stephane" password="stephane74"
C:\Users\Stephane\.node-red\mcu-plugin-cache\configytvokt5ta5i>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'
IDF_TOOLS_PATH not set. Setting to C:\Espressif\
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3
Adding ESP-IDF tools to PATH...
ERROR: Platform Windows- appears to be unsupported
1 file copiati.
1 file copiati.
1 file copiati.
Reconfiguring ESP-IDF...
Loading defaults file C:/Users/Stephane/Projects/MODDABLE/build/tmp/esp32/moddable_two/debug/configytvokt5ta5i/xsProj-esp32/sdkconfig.mc...
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Stephane/Projects/MODDABLE/build/tmp/esp32/moddable_two/debug/configytvokt5ta5i/xsProj-esp32/build
Executing action: reconfigure
Running cmake in directory c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsproj-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=C:/Users/Stephane/Projects/moddable/build/tmp/esp32/moddable_two/debug/configytvokt5ta5i/xsProj-esp32/sdkconfig.mc -DSDKCONFIG_HEADER=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsProj-esp32\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 -DIDF_TARGET=esp32 -DESP32_SUBCLASS=esp32 -DSDKCONFIG_DEFAULTS=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsProj-esp32\sdkconfig.mc -DCCACHE_ENABLE=0 c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsproj-esp32"...
1 file copiati.
# cc xsHost.o (strings in flash)
# cc xsHosts.o (strings in flash)
# cc xsPlatform.o (strings in flash)
C:\Users\Stephane\AppData\Local\Temp\ccP5uA8j.s: Assembler messages:
C:\Users\Stephane\AppData\Local\Temp\ccP5uA8j.s:2589: Warning: setting incorrect section attributes for .rodata.mod.0
# cc xsAll.o (strings in flash)
# cc xsAPI.o (strings in flash)
# cc xsArguments.o (strings in flash)
# cc xsArray.o (strings in flash)
# cc xsAtomics.o (strings in flash)
# cc xsBigInt.o (strings in flash)
# cc xsBoolean.o (strings in flash)
# cc xsCode.o (strings in flash)
# cc xsCommon.o (strings in flash)
# cc xsDataView.o (strings in flash)
# cc xsDate.o (strings in flash)
# cc xsDebug.o (strings in flash)
# cc xsError.o (strings in flash)
# cc xsFunction.o (strings in flash)
# cc xsGenerator.o (strings in flash)
# cc xsGlobal.o (strings in flash)
# cc xsJSON.o (strings in flash)
# cc xsLexical.o (strings in flash)
# cc xsMapSet.o (strings in flash)
# cc xsMarshall.o (strings in flash)
# cc xsMath.o (strings in flash)
# cc xsMemory.o (strings in flash)
# cc xsModule.o (strings in flash)
# cc xsNumber.o (strings in flash)
# cc xsObject.o (strings in flash)
[... 915 lines cut ...]
ng C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/60153870dc3c7799842de2625877
06f5/components/mbed
tls/port/sha/parallel_
engine/esp_sha1.c.obj
[156/1035] Buil
ding C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/e3deb331
a5419fd41c0de1690b2df
98d/mbedtls/port/sha/
parallel_engine/esp_sha
512.c.obj
ninja: bui
ld stopped: subcommand failed.
ninja failed with exit code 1
Executing action: all (aliases: build)
Running cmake in directory c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsproj-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DINSTRUMENT=1 -DTMP_DIR=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i -DmxDebug=1 -DSDKCONFIG_HEADER=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsProj-esp32\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 -DESP32_SUBCLASS=esp32 -DSDKCONFIG_DEFAULTS=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsProj-esp32\sdkconfig.mc -DCCACHE_ENABLE=0 c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsproj-esp32"...
Running ninja in directory c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\configytvokt5ta5i\xsproj-esp32\build
Executing "ninja all"...
Impossibile trovare il file specificato.
Impossibile trovare il file specificato.
1 file copiati.
Impossibile trovare il file specificato.
1 file copiati.
Impossibile trovare il file specificato.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...

ralphwetzel, 2023-02-05:
Could you please as well update your installation of node-red-mcu-plugin? There happened a lot in the last weeks, even after the latest formal release that is available on npm. So, please do a 

```
cd \Users\Stephane\.node-red
npm install http://www.github.com/ralphwetzel/node-red-mcu-plugin
```

This could / should fix the  

```
ERROR: Platform Windows- appears to be unsupported
```

topic - and bring you one step further to a successful build run.

sleveel, 2023-02-05:
Hi,
My node red has stopped working now...
No debug messages and no dashboard.

When I start it I get some new messages:

Welcome to Node-RED
===================

5 Feb 15:40:09 - [info] Node-RED version: v3.0.2
5 Feb 15:40:09 - [info] Node.js  version: v16.17.1
5 Feb 15:40:09 - [info] Windows_NT 10.0.22000 x64 LE
5 Feb 15:40:10 - [info] Loading palette nodes
5 Feb 15:40:10 - [info] Node-RED MCU Edition Runtime Version: #743876a
5 Feb 15:40:10 - [info] Node-RED MCU Edition Plugin  Version: v1.1.2-post
5 Feb 15:40:11 - [info] Moddable SDK Version: vOS221101-50-g76ae9d8 (x86)
5 Feb 15:40:11 - [info] *** node-red-mcu-plugin:
5 Feb 15:40:11 - [info] It looks as if a platform option has been removed.
5 Feb 15:40:11 - [info] Please raise an issue @ our GitHub repository, stating the following information:
5 Feb 15:40:11 - [info] > Verify platform: esp32/esp32s3_usb
5 Feb 15:40:11 - [info] > Verify platform: esp32/qtpys3
5 Feb 15:40:11 - [info] > Verify platform: esp32/s3_tft_feather
5 Feb 15:40:11 - [info] > Verify platform: esp32/wt32_eth01
5 Feb 15:40:12 - [info] Dashboard version 3.2.3 started at /ui
5 Feb 15:40:12 - [info] Settings file  : C:\Users\Stephane\.node-red\settings.js
5 Feb 15:40:12 - [info] Context store  : 'default' [module=memory]
5 Feb 15:40:12 - [info] User directory : \Users\Stephane\.node-red
5 Feb 15:40:12 - [info] Projects directory: C:\Users\Stephane\.node-red\projects
5 Feb 15:40:12 - [info] Server now running at http://127.0.0.1:1880/
5 Feb 15:40:12 - [info] Active project : sideBoardingLadder
5 Feb 15:40:12 - [info] Flows file     : C:\Users\Stephane\.node-red\projects\sideBoardingLadder\flows.json
5 Feb 15:40:12 - [warn] Using unencrypted credentials
5 Feb 15:40:12 - [info] Starting flows
5 Feb 15:40:12 - [info] Started flows

sleveel, 2023-02-05:
Hi, It looks like I got Node Red up and running again.

I am trying to deploy a flow with a UI node but I get this message:

Starting build process...
Host system check: Windows 10 Home
MCU Build system check: p1.1.2-post + #743876a @ mOS221101-50-g76ae9d8 (x86)
HOME directory check: C:\Users\Stephane
Creating build environment for platform esp32/moddable_two.
Error: This flow uses UI nodes - yet UI support is diabled. Please enable UI support.

Can I enable UI support?
Thanks!

sleveel, 2023-02-05:
After updating everything and ùwhen trying to load a simple flow (with no UI) I get:

Starting build process...
Host system check: Windows 10 Home
MCU Build system check: p1.1.2-post + #743876a @ mOS221101-50-g76ae9d8 (x86)
HOME directory check: C:\Users\Stephane
Creating build environment for platform esp32/moddable_two.
Working directory: \Users\Stephane\.node-red\mcu-plugin-cache\ytvokt5ta5i
$IDF_PATH is defined: C:\Espressif\frameworks\esp-idf-v4.4.3
$IDF_TOOLS_PATH is defined: C:\Espressif
$IDF_PYTHON_ENV_PATH is not defined.
> cd \Users\Stephane\.node-red\mcu-plugin-cache\ytvokt5ta5i
UPLOAD_PORT = COM3
Creating build batch file...
> cmd.exe build.bat

C:\Users\Stephane\.node-red\mcu-plugin-cache\ytvokt5ta5i>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat" 
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.24.202208181.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.3\tools
    
Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.3\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

FC: impossibile aprire C:\USERS\STEPHANE\PROJECTS\MODDABLE\BUILD\TMP\ESP32\MODDABLE_TWO\DEBUG\YTVOKT5TA5I\XSPROJ-ESP32\PARTITIONS.CSV - File o directory non esistenti

        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
Reconfiguring ESP-IDF...
Loading defaults file C:/Users/Stephane/Projects/MODDABLE/build/tmp/esp32/moddable_two/debug/ytvokt5ta5i/xsProj-esp32/sdkconfig.mc...
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Stephane/Projects/MODDABLE/build/tmp/esp32/moddable_two/debug/ytvokt5ta5i/xsProj-esp32/build
Executing action: reconfigure
Running cmake in directory c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsproj-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=C:/Users/Stephane/Projects/moddable/build/tmp/esp32/moddable_two/debug/ytvokt5ta5i/xsProj-esp32/sdkconfig.mc -DSDKCONFIG_HEADER=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsProj-esp32\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 -DIDF_TARGET=esp32 -DESP32_SUBCLASS=esp32 -DSDKCONFIG_DEFAULTS=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsProj-esp32\sdkconfig.mc -DCCACHE_ENABLE=1 c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsproj-esp32"...
        1 file copiati.
[... 1554 lines cut ...]
Wrote 1364320 bytes (825784 compressed) at 0x00010000 in 13.4 seconds (effective 812.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 123...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (123 compressed) at 0x00008000 in 0.1 seconds (effective 404.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: all (aliases: build)
Running cmake in directory c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsproj-esp32\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DINSTRUMENT=1 -DTMP_DIR=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i -DmxDebug=1 -DSDKCONFIG_HEADER=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsProj-esp32\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 -DESP32_SUBCLASS=esp32 -DSDKCONFIG_DEFAULTS=C:\Users\Stephane\Projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsProj-esp32\sdkconfig.mc -DCCACHE_ENABLE=1 c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsproj-esp32"...
Running ninja in directory c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsproj-esp32\build
Executing "ninja all"...
Executing action: flash
Running ninja in directory c:\users\stephane\projects\moddable\build\tmp\esp32\moddable_two\debug\ytvokt5ta5i\xsproj-esp32\build
Executing "ninja flash"...
Done
        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
Impossibile trovare il file specificato.
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...

sleveel, 2023-02-05:
But it looks like it worked since Xbug shows the flow running.

ralphwetzel, 2023-02-05:
> But it looks like it worked since Xbug shows the flow running.

Great, that looks like it a successful build process... !! 👍 

> Can I enable UI support?

Yes, you can! This is an option selectable in the Configuration Panel of the plugin:

[image]

In case you don't see this option, click on `Show more options...`

[image]

to display it (and others as well).