ralphwetzel/node-red-mcu-plugin #16: requested USB port not used on ESP8266 builds

phoddie, 2023-02-14:
Setting the USB port works for ESP32 builds (nice!) but seems to be ignored for ESP8266 builds. Here's how I'm configuring it:

[image]

The ESP32 build will try to use the port, but the ESP8266 build always uses the default. I checked the `mcconfig` [scripts for ESP8266](https://github.com/Moddable-OpenSource/moddable/blob/cc39b653e5f2e0e29bd99e00cd5d08bbc68208b5/tools/mcconfig/make.esp.mk#L52-L62) and they do respect the `UPLOAD_PORT` environment variable.

This is on macOS Ventura running the current version of the MCU Plugin.

ralphwetzel, 2023-02-14:
Thank you for reporting.
I've committed a tiny fix; would be great if you find time to give it a try...

phoddie, 2023-02-14:
That looks to be fixed. Thanks for the quick attention.

closed 2023-02-14 by phoddie