ralphwetzel/node-red-mcu-plugin #17: ESP8266 target crashing on startup after connecting to wifi

colinl, 2023-02-14:
Following on from https://github.com/phoddie/node-red-mcu/issues/83 and https://discourse.nodered.org/t/node-red-mcu-plugin-v1-0-integrating-node-red-mcu-edition-into-the-node-red-editor/72693/91

Using @ralphwetzel/node-red-mcu-plugin installed from git #cf57d3d7f0ebac26285bba41b2148d60cbb32446, building a simple flow for a Wemos D1 Mini (configured in the plugin as a Nodemcu), the target crashes after connecting to the wifi.

In node-red-mcu/setupwifi.js in the code

		   case WiFi.gotIP:
				trace(`IP address ${Net.get("IP")}\n`);

				if (!sntp || (Date.now() > 1672722071_000)) {
					done?.();
					done = undefined;
					return;
				}
If I put a breakpoint on the line done?.() it breaks ok and if I step then it moves forwards to the next line, and if I let it run then the flow runs correctly. However, if I put the breakpoint instead on done = undefined then it shows the IP address in the log but then immediately crashes and restarts. At the suggestion of @ralphwetzel in [this forum thread ](https://discourse.nodered.org/t/node-red-mcu-plugin-v1-0-integrating-node-red-mcu-edition-into-the-node-red-editor/72693/101)I hid the esp entry in @ralphwetzel/node-red-mcu-plugin/node-red-mcu/manifest_core.json so that it used legacy code, and rebuilt, but the behaviour was the same.

This seems to happen if I have any contrib nodes installed, but without those it is ok. This simple example uses node-red-contrib-pid

[{"id":"a74a29ab51318a3a","type":"inject","z":"f92bb64b93bfc46e","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"5","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","_mcu":{"mcu":true},"x":170,"y":140,"wires":[["9db682e1f6a487a8"]]},{"id":"9db682e1f6a487a8","type":"debug","z":"f92bb64b93bfc46e","name":"debug 80","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","_mcu":{"mcu":true},"x":370,"y":200,"wires":[]},{"id":"6d9adcf8cdc36b27","type":"debug","z":"f92bb64b93bfc46e","name":"debug 81","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","_mcu":{"mcu":true},"x":440,"y":300,"wires":[]},{"id":"ed0d63aad317e829","type":"inject","z":"f92bb64b93bfc46e","name":"","props":[{"p":"payload"}],"repeat":"2","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"1","payloadType":"num","_mcu":{"mcu":true},"x":90,"y":380,"wires":[["a83bf3122efb505c"]]},{"id":"a83bf3122efb505c","type":"trigger","z":"f92bb64b93bfc46e","name":"","op1":"0","op2":"1","op1type":"num","op2type":"num","duration":"500","extend":false,"overrideDelay":false,"units":"ms","reset":"","bytopic":"all","topic":"topic","outputs":1,"_mcu":{"mcu":true},"x":280,"y":380,"wires":[["6d9adcf8cdc36b27","32dcd5abbb576d04"]]},{"id":"7180d313e18f3785","type":"debug","z":"f92bb64b93bfc46e","name":"debug 82","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","_mcu":{"mcu":true},"x":440,"y":480,"wires":[]},{"id":"32dcd5abbb576d04","type":"rpi-gpio out","z":"f92bb64b93bfc46e","name":"","pin":"2","set":"","level":"0","freq":"","out":"out","bcm":true,"_mcu":{"mcu":true},"x":450,"y":380,"wires":[]},{"id":"e0f56ffc112362dd","type":"inject","z":"f92bb64b93bfc46e","name":"","props":[{"p":"payload"}],"repeat":"2","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"0.5","payloadType":"num","_mcu":{"mcu":true},"x":90,"y":480,"wires":[[]]},{"id":"b74829da6239b15d","type":"PID","z":"f92bb64b93bfc46e","name":"","setpoint":21,"pb":1,"ti":9999,"td":0,"integral_default":0.5,"smooth_factor":3,"max_interval":600,"enable":1,"disabled_op":0,"_mcu":{"mcu":true},"x":220,"y":580,"wires":[[]]}]

Attached are zips of the environment using the plugin and the environment building the same flow using node-red-mcu directly.
[using-node-red-mcu.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10733542/using-node-red-mcu.zip)
[using-plugin-xbu2qex629n.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10733589/using-plugin-xbu2qex629n.zip)

ralphwetzel, 2023-02-14:
Thank you for that report. I've an idea of what could cause this.
A D1 Mini is on its way to me & should arrive at the end of the week.
Keep you updated...

ralphwetzel, 2023-02-17:
@colinl 
As I suspected: The cause for your trouble is #19.
Sorry that you had to spend so much time to investigate for a change of a simple memory layout definition setting. Enough said...

closed 2023-02-17 by ralphwetzel

colinl, 2023-02-18:
No problem, when one tries something new one expects to find teething problems.
I can confirm that it is fixed for me too.

commit 72edb36f (2023-02-17): Use manifest_host.json

Closes #17