ralphwetzel/node-red-mcu-plugin #20: [FR] Include handling

Sineos, 2023-02-15:
I have learned that the sensor nodes from @phoddie need to include their respective drivers as well.
For example the FT6206:

```
  "include": [
    ...
	"$(MODDABLE)/modules/drivers/sensors/ft6206/manifest.json"
  ],
```

it would be awesome, if the MCU-plugin could collect these paths from https://github.com/phoddie/node-red-mcu/blob/main/nodes/sensor/resources/sensors.json and include automatically. 

Alternatively a way to manage includes from within the plugin would be a great help as well.

In any case thanks a lot to you and @phoddie for your work and commitment. Really great and highly appreciated.

ralphwetzel, 2023-02-15:
Well... that's a bug in those nodes.
The plugin provides an interface for any node to register additional dependencies. It's a 3-liner.
@phoddie [is well aware of that...](https://github.com/phoddie/node-red-mcu/pull/71)

Sineos, 2023-02-15:
Thanks for the link. I understand that the best approach is still under discussion. Fair enough