ralphwetzel/node-red-mcu-plugin #21: Problems with Moddable Placement on Windows

NextWorldLab, 2023-02-16:
If you put Moddable in D drive in Windows environment and start Node-RED from C drive, the build script will not work.There seems to be a problem in lines 271-275 of mcu_plugin.js, but I don't know what to do.
I think that xsbug cannot start because the Windows cd does not change the drive.

```
           `CALL "${process.env["ProgramFiles"]}\\Microsoft Visual Studio\\2022\\Community\\VC\\Auxiliary\\Build\\vcvars64.bat" > nul`,
            `cd ${MODDABLE}\\build\\bin\\win\\debug`,
            'dumpbin /headers xsbug.exe | findstr "machine"'
```

Sineos, 2023-02-16:
Your theory sounds possible. You can test it by replacing

https://github.com/ralphwetzel/node-red-mcu-plugin/blob/3f3aec6ee74b671cf2c07b4a4a18446d3533be79/mcu_plugin.js#L273

with
```
cd /D ${MODDABLE}\\build\\bin\\win\\debug
```

and potentially (although I'm not sure)

https://github.com/ralphwetzel/node-red-mcu-plugin/blob/3f3aec6ee74b671cf2c07b4a4a18446d3533be79/mcu_plugin.js#L1820

with
```
publish_stdout(`> cd /D ${make_dir}\n`); 
```

ralphwetzel, 2023-02-16:
Good catch! Thank you for your report!

I'd need some data to address this:

- Your definition of $MODDABLE
- The first few lines (about 10-15) Node-RED emits to the console when starting.
- A copy of the log messages you get in the `Console Monitor` (plugin panel) when you run a build.

When posting data here, please ensure to remove sensitive information (e.g. passwords) before...

NextWorldLab, 2023-02-16:
MODDABLE environment variable：”d:\github\moddable”

Node-RED Start Message
```
C:\Users\*****>node-red
16 Feb 16:41:32 - [info]

Welcome to Node-RED
===================

16 Feb 16:41:32 - [info] Node-RED version: v3.0.2
16 Feb 16:41:32 - [info] Node.js  version: v18.13.0
16 Feb 16:41:32 - [info] Windows_NT 10.0.22621 x64 LE
16 Feb 16:41:32 - [info] Loading palette nodes
16 Feb 16:41:33 - [info] Node-RED MCU Edition Runtime Version: #8224a11
16 Feb 16:41:33 - [info] Node-RED MCU Edition Plugin  Version: v1.1.2-post
16 Feb 16:41:34 - [info] Moddable SDK Version: v82c864a
16 Feb 16:41:34 - [info] *** node-red-mcu-plugin:
16 Feb 16:41:34 - [info] It looks as if a new platform option has been added.
16 Feb 16:41:34 - [info] Please raise an issue @ our GitHub repository, stating the following information:
16 Feb 16:41:34 - [info] > New platform: esp32/atoms3
16 Feb 16:41:34 - [info] > New platform: esp32/atoms3_lite
16 Feb 16:41:34 - [info] Dashboard version 3.3.1 started at /ui
16 Feb 16:41:34 - [info] Settings file  : C:\Users\*****\.node-red\settings.js
16 Feb 16:41:34 - [info] Context store  : 'default' [module=memory]
16 Feb 16:41:34 - [info] User directory : \Users\*****\.node-red
16 Feb 16:41:34 - [warn] Projects disabled : editorTheme.projects.enabled=false
16 Feb 16:41:34 - [warn] Flows file name not set. Generating name using hostname.
16 Feb 16:41:34 - [info] Flows file     : \Users\*****\.node-red\flows_t-frontier.json
16 Feb 16:41:34 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

16 Feb 16:41:34 - [warn] Encrypted credentials not found
16 Feb 16:41:34 - [info] Server now running at http://127.0.0.1:1880/
16 Feb 16:41:34 - [info] Starting flows
16 Feb 16:41:34 - [info] Started flows
```

Build Message
```
Starting build process...
Host system check: Windows 10 Pro
MCU Build system check: p1.1.2-post + #8224a11 @ m82c864a
HOME directory check: C:\Users\*****
Creating build environment for platform esp32/m5stick_cplus.
Working directory: \Users\*****\.node-red\mcu-plugin-cache\ilk6j6qziz
$IDF_PATH is defined: C:\Espressif\frameworks\esp-idf-v4.4.4
$IDF_TOOLS_PATH is defined: C:\Espressif
$IDF_PYTHON_ENV_PATH is not defined.
> cd \Users\*****\.node-red\mcu-plugin-cache\ilk6j6qziz
UPLOAD_PORT = undefined
Creating build batch file...
> cmd.exe build.bat

C:\Users\tauchi\.node-red\mcu-plugin-cache\ilk6j6qziz>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvarsundefined.bat" 
'"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvarsundefined.bat"' �́A�����R�}���h�܂��͊O���R�}���h�A
����\�ȃv���O�����܂��̓o�b�` �t�@�C���Ƃ��ĔF������Ă��܂���B

C:\Users\*****\.node-red\mcu-plugin-cache\ilk6j6qziz>pushd C:\Espressif\frameworks\esp-idf-v4.4.4 

C:\Espressif\frameworks\esp-idf-v4.4.4>CALL "C:\Espressif\idf_cmd_init.bat" 
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.4

Adding ESP-IDF tools to PATH...
    C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.4\tools
    
Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.4\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Espressif\tools\idf-exe\1.0.3\idf.py.exe
C:\Espressif\frameworks\esp-idf-v4.4.4\tools\idf.py
### Cannot execute nmake!
```

Booting from C drive gives an error, booting from D drive is OK.

thank you,

NextWorldLab, 2023-02-16:
The /D switch looks good, so I'll make a pull request.

ralphwetzel, 2023-02-16:
We should keep this open - until confirmed to be solved...

closed 2023-02-17 by ralphwetzel