ralphwetzel/node-red-mcu-plugin #24: ESP8266 flow runs on node-red-mcu standalone, but not via plugin

colinl, 2023-02-20:
The attached flow runs perfectly when built in node-red-mcu standalone, but crashes after connecting to the wifi and running (corrrectly) for a short time when run via the plugin.
Running in an ESP8266 Wemos D1 Mini configured as a nodemcu.
They are both using the same install of moddable and node-red-mcu is the current version pulled from github in both cases.
The plugin is the version from github a day or two ago,  #72edb36f317e...

I wonder whether there is a difference in the memory configuration.  At the moment of the crash in the plugin I see
System bytes free 4456 bytes
Chunk used 4152 / 4960 bytes
Slot used 11024 / 12272 bytes
Stack used 2400 /6144 bytes

Whereas at about the same time after startup, using node-red-mcu standalone I see
System bytes free 4456
Chunk used 12804 / 13044 bytes
Slot used 11248 / 11248 bytes
Stack used 2400 /6144 bytes

When built standalone I use 
`mcconfig -d -m -p esp/nodemcu ssid="**" password="**"`

The flow requires node-red-contrib-pid and node-red-contrib-timeprop.
```
[{"id":"ca71fe5f83dd3ed0","type":"tab","label":"mcu test","disabled":false,"info":"","env":[],"_mcu":{"mcu":true}},{"id":"e7557c4401194004","type":"debug","z":"ca71fe5f83dd3ed0","name":"debug 85","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","_mcu":{"mcu":true},"x":900,"y":100,"wires":[]},{"id":"daf71fd6976e6abe","type":"mqtt out","z":"ca71fe5f83dd3ed0","name":"test/mcu/pid/result","topic":"test/mcu/pid/result","qos":"0","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"75fb29423f8a0770","_mcu":{"mcu":true},"x":590,"y":260,"wires":[]},{"id":"f1a74036d852a399","type":"mqtt in","z":"ca71fe5f83dd3ed0","name":"","topic":"test/mcu/pid/set/#","qos":"1","datatype":"json","broker":"75fb29423f8a0770","nl":false,"rap":true,"rh":0,"inputs":0,"_mcu":{"mcu":true},"x":100,"y":80,"wires":[["e7c1c7cd2ed2ef01"]]},{"id":"e7c1c7cd2ed2ef01","type":"function","z":"ca71fe5f83dd3ed0","name":"Set topic","func":"msg.topic =  msg.topic.split(\"/\").slice(-1)[0]\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"_mcu":{"mcu":true},"x":140,"y":180,"wires":[["d9270432a6668b06"]]},{"id":"d9270432a6668b06","type":"PID","z":"ca71fe5f83dd3ed0","name":"","setpoint":21,"pb":"10","ti":9999,"td":0,"integral_default":0.5,"smooth_factor":3,"max_interval":600,"enable":1,"disabled_op":0,"_mcu":{"mcu":true},"x":290,"y":180,"wires":[["c3d5fbd10c6676dc","e8104385c3f36513"]]},{"id":"c3d5fbd10c6676dc","type":"timeprop","z":"ca71fe5f83dd3ed0","name":"","cycleTime":"10","deadTime":0,"triggerPeriod":"1","invert":false,"_mcu":{"mcu":true},"x":440,"y":180,"wires":[["48287ccabfb7e55a","daf71fd6976e6abe"]]},{"id":"e8104385c3f36513","type":"debug","z":"ca71fe5f83dd3ed0","name":"debug 86","active":true,"tosidebar":false,"console":true,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","_mcu":{"mcu":true},"x":510,"y":60,"wires":[]},{"id":"48287ccabfb7e55a","type":"rpi-gpio out","z":"ca71fe5f83dd3ed0","name":"LED","pin":"2","set":"","level":"0","freq":"","out":"out","bcm":true,"_mcu":{"mcu":true},"x":610,"y":180,"wires":[]},{"id":"36501d946e21bc46","type":"rpi-ds18b20","z":"ca71fe5f83dd3ed0","topic":"","array":false,"name":"","_mcu":{"mcu":true},"x":270,"y":260,"wires":[[]]},{"id":"5da2de6f6e27f039","type":"inject","z":"ca71fe5f83dd3ed0","name":"Measure","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","_mcu":{"mcu":true},"x":100,"y":260,"wires":[[]]},{"id":"75fb29423f8a0770","type":"mqtt-broker","name":"Owl2 for mcu","broker":"192.168.49.83","port":"1883","clientid":"","autoConnect":true,"usetls":false,"protocolVersion":"4","keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","birthMsg":{},"closeTopic":"","closeQos":"0","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willPayload":"","willMsg":{},"userProps":"","sessionExpiry":"","_mcu":{"mcu":false}}]
```

ralphwetzel, 2023-02-20:
The memory configuration is defined via the `manifest.json` created by the plugin.
Would you mind posting this here as well?

colinl, 2023-02-20:
Do you mean the one in mcu-plugin-cache/xbu2qex629n? If so I have attached it obfuscated to take out the wifi credentials.  In case you meant manifest_host from the plugin I have included that too.
[manifests.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10786678/manifests.zip)

colinl, 2023-02-20:
I realise that the manifest_core I posted is actually from node-red-mcu under the plugin, so not sure which file you want.

ralphwetzel, 2023-02-20:
Thank you for the manifest.json.
I get exactly the same when building your flow.

```
{
  "build": {
    "MODULES": "/home/colinl/Projects/moddable/modules",
    "MCUROOT": "/home/colinl/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu",
    "REDNODES": "/usr/lib/node_modules/node-red/node_modules/@node-red/nodes"
  },
  "include": [
    "$(MCUROOT)/manifest_host.json",
    "/home/colinl/.node-red/mcu-plugin-cache/xbu2qex629n/node-red-contrib-pid/manifest.json",
    "/home/colinl/.node-red/mcu-plugin-cache/xbu2qex629n/node-red-contrib-timeprop/manifest.json",
    "$(MCUROOT)/nodes/rpi-gpio/manifest.json",
    "$(MCUROOT)/nodes/rpi-ds18b20/manifest.json"
  ],
  "modules": {
    "*": [
      "./main",
      {
        "source": "./flows",
        "transform": "nodered2mcu"
      }
    ],
    "~": []
  },
  "config": {
    "ssid": "**",
    "password": "**",
    "noderedmcu": {
      "editor": true
    }
  }
}
```

As you can see, there's no modification of the memory management at all & the rest is straight & valid `node-red-mcu`.

How did you setup the other part of your test?
There needs to be a difference... ?

colinl, 2023-02-20:
I just copy the flows file across to the Projects folder where node-red-mcu installed, copy the two contrib nodes across to the nodes folder there, setup the manifest files for those, and build it.

ralphwetzel, 2023-02-20:
> setup the manifest files for those

Is there a difference to the manifests the plugin creates in 

```
    "/home/colinl/.node-red/mcu-plugin-cache/xbu2qex629n/node-red-contrib-pid/manifest.json",
    "/home/colinl/.node-red/mcu-plugin-cache/xbu2qex629n/node-red-contrib-timeprop/manifest.json",
```

What happens, when you set `editor: false` in the plugin-created `manifest.json`:

```
    "noderedmcu": {
      "editor": true
    }
```

colinl, 2023-02-20:
In your node-red-contrib-pid/manifest.json you have
```
{
  "//": {
    "***": "https://github.com/ralphwetzel/node-red-mcu-plugin",
    "npm": "node-red-contrib-pid",
    "xs": "manifest.json",
    "@": "2023-02-20T21:04:32.166Z",
    "ref": "https://github.com/Moddable-OpenSource/moddable"
  },
  "build": {
    "MCUMODULES": "/home/colinl/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/mcu_modules",
    "REDNODES": "/usr/lib/node_modules/node-red/node_modules/@node-red/nodes"
  },
  "include": [
    "$(MCUMODULES)/require/manifest.json"
  ],
  "modules": {
    "*": [],
    "node-red-contrib-pid": "/home/colinl/.node-red/node_modules/node-red-contrib-pid/pid"
  }
```
Whereas all I have in mine is
```
{
	"modules": {
		"*": "./pid"
	},
	"preload": "pid"
}
```
That file is inside node-red-mcu/nodes/node-red-contrib-pid.  Then the main manifest includes that one.

ralphwetzel, 2023-02-20:
Thank you for posting the data.
I can spot two things that are - technically - different here:

- The plugin `includes` the `require` module.
- The plugin does not `preload` the `modules` - in this case `pid`.

You may verify if aligning these details has an impact on the behavior you see...

colinl, 2023-02-21:
I modified the two contrib node manifest files to contain just
```
{
  "modules": {
    "*": [],
    "node-red-contrib-pid": "/home/colinl/.node-red/node_modules/node-red-contrib-pid/pid"
  },
	"preload": "pid"
}
```
and
```
{
  "modules": {
    "*": [],
    "node-red-contrib-timeprop": "/home/colinl/.node-red/node_modules/node-red-contrib-timeprop/timeprop"
  },
	"preload": "timeprop"
}
```
and it still crashed in the same way.
Like you, I cannot see anything in the manifests that might cause this.

To make sure this is not a hangover from previous tests, I moved away the .node-red folder and restarted node-red so that it recreated the folder with no flows and no installed nodes.  I installed the necessary nodes, and installed the plugin direct from github, and imported the flow from above, built it and it crashes as before.

I then double checked that I am using the same flow file in ~/Projects/node-red-mcu, cleaned and built it and it performs as it should.

Did your D1 Mni arrive?  If so are you able to replicate the issue?

colinl, 2023-02-22:
Right, got it.
In manifest.json you need
```
  "preload": [
        "flows"
  ],
```
I am guessing that has some effect on the initial distribution of RAM, reducing fragmentation or something.  Probably also a good idea to include `preload` that in the contrib nodes manifest files too.

ralphwetzel, 2023-02-23:
Thank you for catching this.
I've changed the default to always preload "flows" now. The patch is already in the repo...
I'll test, if this is suitable for all contrib nodes as well...

colinl, 2023-02-23:
Working well for me, thanks.  That was the last thing stopping me moving over to using the plugin.

Is there a reason you have reservations about preloading contrib nodes?

ralphwetzel, 2023-02-23:
> Is there a reason you have reservations about preloading contrib nodes?

I don't have any reservations doing this. I just need to ensure it's "the right thing" for any case. As a contrib node is nothing else than a npm package & the plugin processes the whole dependency tree, preloading a contrib node means preloading any npm package. ... or not? Therefore: This needs some further moments to consider the details...

ralphwetzel, 2023-02-23:
After reading [this document](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md), I think it's not a good choice to preload contrib nodes per default:

- To support preloading, a module needs to respect some constraints. This can probably be ensured for code dedicated to run in XS, but not for packages unaware at all of what we're trying to do here...
- As stated [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md#what-cannot-be-preloaded), an error will be emitted in case the preloading fails ... yet this error is far from being self-explanatory.
- Conclusion for me: When there is a positive confirmation that a contrib node can be preloaded, it's easy to provide a templated `manifest.json` that enables preloading.
- Per default, contrib nodes (or any other npm packages) will not be preloaded.

I'm still missing a clever idea how (and if) to support this "testing if a module could be preloaded" from within the plugin...

phoddie, 2023-02-24:
> I'm still missing a clever idea how (and if) to support this "testing if a module could be preloaded" from within the plugin...

It is not practical. Even if you can preflight the build, there is still the potential for runtime failures related to preloaded objects being immutable.

I've yet to see a Node-RED node implementation that cannot run preloaded. I'm sure one exists. Still, the practical reality is that preloading is essential for reducing RAM use on resource constrained device. Enabling preload by default is the pragmatic choice.

ralphwetzel, 2023-03-17:
@colinl : Can we close this issue as resolved?

colinl, 2023-03-17:
Have you decided what you are doing about preloading, or otherwise, contrib nodes?

ralphwetzel, 2023-03-18:
There will be an option to opt-in preloading of contributed nodes.

colinl, 2023-03-18:
Perfect.

closed 2023-03-18 by colinl