ralphwetzel/node-red-mcu-plugin #27: Raise allowed version of node-red?

colinl, 2023-03-04:
Currently package.json specifies
```
  "node-red": {
    "version": ">=2.0.0 <=3.0.2",
``` 
I have hacked that locally to allow 3.1 Beta to be used and (for me at least) it runs ok.
Would it be appropriate to increase the allowed version in package.json?

In fact I suggest it should be ok to allow <4.0.0 as the node-red team is very careful when it comes to backwards compatibility, so I think the chance of this causing problems is small.

mkellner, 2023-09-14:
The default windows install of node-red was v3.1.0. The subsequent install of node-red-mcu-plugin failed:
```
14 Sep 14:56:44 - [warn] [@ralphwetzel/node-red-mcu-plugin] Node module cannot be loaded on this version. Requires: >=2.0.0 <=3.0.2
```
Can you please increase the version?

Thank you

ralphwetzel, 2023-09-26:
v1.4 now supports NR 3.1

closed 2023-09-26 by ralphwetzel