ralphwetzel/node-red-mcu-plugin #29: trigger node

mshioji, 2023-03-10:
Hello, I'm trying plugin with recent released moddable SDK and node-red-mcu.
I got an error when I build the flow with "Trigger" node as follows:
### Error: '/home/mshioji/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/nodes/trigger/manifest.json': manifest not found! 
the error shows manifest.json is in the :  node-red-mcu-plugin/node-red-mcu/nodes/trigger/
but now it is here (one layer deeper )  :  node-red-mcu-plugin/node-red-mcu/function/nodes/trigger/
manifest_runtime.json of node-red-mcu points here correctly, so direct build from node-red mcu works fine.
I tried re-install twice on Ubuntu, then Windows, all the same situation.
I'm not sure but is there a place that previous module path still alive in the plugin code? 
Thank you,

ralphwetzel, 2023-03-10:
Thank you for reporting this issue.
You may have noticed, that `node-red-mcu` changed it's directory structure a few days ago. The adaptations of the plugin to work with this new structure are in the plugin's repo - yet were not put in a new release.

I took your issue as a trigger to create a [new release](https://github.com/ralphwetzel/node-red-mcu-plugin/releases/tag/v1.2.1). This is on [npm](https://www.npmjs.com/package/@ralphwetzel/node-red-mcu-plugin) as well, so another install run will bring you back in the game...

mshioji, 2023-03-11:
Thank you for the quick release, I could get back in the game. ;)

closed 2023-03-11 by mshioji