ralphwetzel/node-red-mcu-plugin #32: manifest not found!

MatsA, 2023-03-17:
Really a fantastic project !

Doing my first build on a RPi, Bullseye, and Node-Red 3.0.2 I get the error as below.

Checking I found that the mentioned "manifest.json" is instead available at 

/home/pi/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/nodes/rpi/rpi-gpio/manifest.json

Please advice, Mats 

******
Host system check: #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023
MCU Build system check: p1.1.2 + #f2f2c68 @ m3.7.6-8-g746d137
HOME directory check: /home/pi
Creating build environment for platform pico/pico_w.
Working directory: /home/pi/.node-red/mcu-plugin-cache/tns8c0j9h9b
$PICO_GCC_ROOT identified: /usr
$PICO_SDK_DIR identified: /home/pi/pico/pico-sdk
> cd /home/pi/.node-red/mcu-plugin-cache/tns8c0j9h9b
Creating build script file...
> /bin/bash ./build.sh
### Error: '/home/pi/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/nodes/rpi-gpio/manifest.json': manifest not found!
********

ralphwetzel, 2023-03-17:
Hi!
Thank you for your feedback & for reporting this issue.

> MCU Build system check: p1.1.2 + #f2f2c68 @ m3.7.6-8-g746d137

You need to update the plugin. We changed the directory structure of `node-red-mcu` recently & this needs >v1.2 of the plugin to work properly:

```
cd <userdir>
npm install @ralphwetzel/node-red-mcu-plugin
```

This should give you v1.2.1.

MatsA, 2023-03-18:
Thanks for quick response, appreciated. For the moment not at the right "place" will check your advice later next week. Thanks.

MatsA, 2023-03-23:
Now tested and works ....

But had to do an uninstall and then a `npm install https://github.com/ralphwetzel/node-red-mcu-plugin.git` to get the latest.

But another issue appeared https://github.com/ralphwetzel/node-red-mcu-plugin/issues/33#issue-1637573851

closed 2023-03-23 by MatsA