ralphwetzel/node-red-mcu-plugin #34: xsbug not connect

NextWorldLab, 2023-03-26:
hello

After updating moddable and plugins a few days ago, xsbug no longer connects. Since it is not linked, the status is not displayed in the debug side panel or work space of the Node-RED flow editor.

Device Windown10 + M5Stick_cplus(ESP32)
environment
```
26 Mar 14:28:25 - [info] Node-RED version: v3.0.2
26 Mar 14:28:25 - [info] Node.js  version: v18.13.0
26 Mar 14:28:25 - [info] Windows_NT 10.0.19045 x64 LE
26 Mar 14:28:27 - [info] Loading palette nodes
26 Mar 14:28:28 - [info] Node-RED MCU Edition Runtime Version: #942132f
26 Mar 14:28:28 - [info] Node-RED MCU Edition Plugin  Version: v1.3.0
26 Mar 14:28:32 - [info] Moddable SDK Version: v3.7.6-57-g1af75c0 (x86)
26 Mar 14:28:34 - [info] Dashboard version 3.4.0 started at /ui
26 Mar 14:28:35 - [info] Settings file  : C:\Users\tauchi\.node-red\settings.js
26 Mar 14:28:35 - [info] Context store  : 'default' [module=memory]
26 Mar 14:28:35 - [info] User directory : \Users\tauchi\.node-red
```
Node-red　Console(I don't see the xsbug connect message.)
[image]

Flow
```
[
    {
        "id": "aeda8097473d33aa",
        "type": "tab",
        "label": "フロー 1",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "9822954b1666b367",
        "type": "inject",
        "z": "aeda8097473d33aa",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": true
        },
        "x": 120,
        "y": 100,
        "wires": [
            [
                "982b20eee84897ff"
            ]
        ]
    },
    {
        "id": "982b20eee84897ff",
        "type": "debug",
        "z": "aeda8097473d33aa",
        "name": "debug 2",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": true
        },
        "x": 360,
        "y": 80,
        "wires": []
    }
]
```
Thank you,

ralphwetzel, 2023-03-26:
@NextWorldLab 
FYI: The issue should be fixed; confirmation for the Windows platform very welcome!

NextWorldLab, 2023-03-27:
Windows 10 still doesn't work.
I'm watching some more.

thank you,

closed 2023-03-27 by NextWorldLab