ralphwetzel/node-red-mcu-plugin #39: TLS-config credentials not exported

phoddie, 2023-08-03:
With some encouragement from @mshioji, I've got a first step in place to support TLS. It is working for HTTP requests. The details are in [this issue](https://github.com/phoddie/node-red-mcu/issues/120).

We are having a problem getting the certificate when using the plug-in. The TLS-Config editor provides two ways to include the certificate in the flow: referencing the certificate by file path and uploading the certificate to the flow. The check box labeled "Use key and certificates from local files" controls which is used. When the certificate is referenced by file path, everything works as expected. When uploading the certificate, the certificate is not included in the flows_cred_mcu.json file as expected. Based on what Node-RED itself does, it should be in the `cadata` property:

```json
    {
        "id": "c1445cf1202e809f",
        "type": "tls-config",
        "name": "",
        "cert": "",
        "key": "",
        "ca": "",
        "certname": "",
        "keyname": "",
        "caname": "ca29.pem",
        "servername": "",
        "verifyservercert": true,
        "alpnprotocol": "",
        "credentials": {
            "cadata": "-----BEGIN CERTIFICATE-----\nMIIEGjCCAwICEQCbfgZJoz5iudXukEhxKe9XMA0GCSqGSIb3DQEBBQUAMIHKMQsw\nCQYDVQQGEwJVUzEXMBUGA1UEChMOVmVyaVNpZ24sIEluYy4xHzAdBgNVBAsTFlZl\ncmlTaWduIFRydXN0IE5ldHdvcmsxOjA4BgNVBAsTMShjKSAxOTk5IFZlcmlTaWdu\nLCBJbmMuIC0gRm9yIGF1dGhvcml6ZWQgdXNlIG9ubHkxRTBDBgNVBAMTPFZlcmlT\naWduIENsYXNzIDMgUHVibGljIFByaW1hcnkgQ2VydGlmaWNhdGlvbiBBdXRob3Jp\ndHkgLSBHMzAeFw05OTEwMDEwMDAwMDBaFw0zNjA3MTYyMzU5NTlaMIHKMQswCQYD\nVQQGEwJVUzEXMBUGA1UEChMOVmVyaVNpZ24sIEluYy4xHzAdBgNVBAsTFlZlcmlT\naWduIFRydXN0IE5ldHdvcmsxOjA4BgNVBAsTMShjKSAxOTk5IFZlcmlTaWduLCBJ\nbmMuIC0gRm9yIGF1dGhvcml6ZWQgdXNlIG9ubHkxRTBDBgNVBAMTPFZlcmlTaWdu\nIENsYXNzIDMgUHVibGljIFByaW1hcnkgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkg\nLSBHMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAMu6nFL8eB8aHm8b\nN3O9+MlrlBIwT/A2R/XQkQr1F8ilYcEWQE37imGQ5XYgwREGfassbqb1EUGO+i2t\nKmFZpGcmTNDovFJbcCAEWNF6yaRpvIMXZK0Fi7zQWM6NjPXr8EJJC52XJ2cybuGu\nkxUccLwgTS8Y3pKI6GyFVxEa6X7jJhFUokWWVYPKMIno3Nij7SqAP395ZVc+FSBm\nCC+Vk7+qRy+oRpfwEuL+wgorUeZ25rdGt+INpsyow0xZVYnm6FNcHOqd8GIWC6fJ\nXwzw3sJ2zq/3avL6QaaiMxTJ5Xpj055iN9WFZZ4O5lMkdBteHRJTW8cs54NJOxWu\nimi5V5cCAwEAATANBgkqhkiG9w0BAQUFAAOCAQEAERSWwauSCPc/L8my/uRan2Te\n2yFPhpk0djZX3dAVL8WtfxUfN2JzPtTnX84XA9s1+ivbrmAJXx5fj267Cz3qWhMe\nDGBvtcC1IyIuBwvLqXTLR7sdwdela8wv0kL9Sd2nic9TutoAWii/gt/4uhMdUIaC\n/Y4wjylGsB49Ndo4YhYYSq3mtlFs3q9i6wHQHiT+eo8SGhJouPtmmRQURVyu565p\nF4ErWjfJXir0xuKhXFSbplQAz/DxwceYMBo7Nhbbo27q/a2ywtrvAkcTisDxszGt\nTxzhT5yvDwyd93gN2PQ1VoDat20Xj50egWTh/sVFuq1ruQp6Tk9LhO5L8X3dEQ==\n-----END CERTIFICATE-----\n"
        }
    }
```

Would you have a look? Thank you!

ralphwetzel, 2023-08-03:
Walking through the plugins code I see no obvious reason why this particular case should / could fail.
Could you - or perhaps @mshioji - please provide a flow that I can use to verify what's happening?
Thank you in advance!

phoddie, 2023-08-03:
Thanks for taking a look. The challenge in providing a flow is that the credentials are stripped on export. I've manually put together this flow that should allow you to see the problem that @mshioji reports.

<details>
<summary>flow.json</summary>

```json
[
    {
        "id": "7d0d6d4406798076",
        "type": "tab",
        "label": "Flow 4",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": false
        }
    },
    {
        "id": "e35ea5b35fd6db16",
        "type": "http request",
        "z": "7d0d6d4406798076",
        "name": "",
        "method": "GET",
        "ret": "txt",
        "paytoqs": "ignore",
        "url": "www.moddable.com/press",
        "tls": "fbdc49382f0f04fa",
        "persist": false,
        "proxy": "",
        "insecureHTTPParser": false,
        "authType": "",
        "senderr": false,
        "headers": [],
        "_mcu": {
            "mcu": false
        },
        "credentials": {},
        "x": 330,
        "y": 140,
        "wires": [
            [
                "d412213eb76c429a"
            ]
        ]
    },
    {
        "id": "d412213eb76c429a",
        "type": "debug",
        "z": "7d0d6d4406798076",
        "name": "debug 87",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": false
        },
        "x": 540,
        "y": 140,
        "wires": []
    },
    {
        "id": "ec26cb026ae62900",
        "type": "inject",
        "z": "7d0d6d4406798076",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": false
        },
        "x": 130,
        "y": 140,
        "wires": [
            [
                "e35ea5b35fd6db16"
            ]
        ]
    },
    {
        "id": "fbdc49382f0f04fa",
        "type": "tls-config",
        "name": "",
        "cert": "",
        "key": "",
        "ca": "",
        "certname": "",
        "keyname": "",
        "caname": "ca109.pem",
        "servername": "",
        "verifyservercert": true,
        "alpnprotocol": "",
        "_mcu": {
            "mcu": false
        },
        "credentials": {
            "cadata": "-----BEGIN CERTIFICATE-----\nMIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQYJKoZIhvcNAQELBQAw\nTzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh\ncmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4\nWhcNMzUwNjA0MTEwNDM4WjBPMQswCQYDVQQGEwJVUzEpMCcGA1UEChMgSW50ZXJu\nZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgUm9vdCBY\nMTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygc\nh77ct984kIxuPOZXoHj3dcKi/vVqbvYATyjb3miGbESTtrFj/RQSa78f0uoxmyF+\n0TM8ukj13Xnfs7j/EvEhmkvBioZxaUpmZmyPfjxwv60pIgbz5MDmgK7iS4+3mX6U\nA5/TR5d8mUgjU+g4rk8Kb4Mu0UlXjIB0ttov0DiNewNwIRt18jA8+o+u3dpjq+sW\nT8KOEUt+zwvo/7V3LvSye0rgTBIlDHCNAymg4VMk7BPZ7hm/ELNKjD+Jo2FR3qyH\nB5T0Y3HsLuJvW5iB4YlcNHlsdu87kGJ55tukmi8mxdAQ4Q7e2RCOFvu396j3x+UC\nB5iPNgiV5+I3lg02dZ77DnKxHZu8A/lJBdiB3QW0KtZB6awBdpUKD9jf1b0SHzUv\nKBds0pjBqAlkd25HN7rOrFleaJ1/ctaJxQZBKT5ZPt0m9STJEadao0xAH0ahmbWn\nOlFuhjuefXKnEgV4We0+UXgVCwOPjdAvBbI+e0ocS3MFEvzG6uBQE3xDk3SzynTn\njh8BCNAw1FtxNrQHusEwMFxIt4I7mKZ9YIqioymCzLq9gwQbooMDQaHWBfEbwrbw\nqHyGO0aoSCqI3Haadr8faqU9GY/rOPNk3sgrDQoo//fb4hVC1CLQJ13hef4Y53CI\nrU7m2Ys6xt0nUW7/vGT1M0NPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNV\nHRMBAf8EBTADAQH/MB0GA1UdDgQWBBR5tFnme7bl5AFzgAiIyBpY9umbbjANBgkq\nhkiG9w0BAQsFAAOCAgEAVR9YqbyyqFDQDLHYGmkgJykIrGF1XIpu+ILlaS/V9lZL\nubhzEFnTIZd+50xx+7LSYK05qAvqFyFWhfFQDlnrzuBZ6brJFe+GnY+EgPbk6ZGQ\n3BebYhtF8GaV0nxvwuo77x/Py9auJ/GpsMiu/X1+mvoiBOv/2X/qkSsisRcOj/KK\nNFtY2PwByVS5uCbMiogziUwthDyC3+6WVwW6LLv3xLfHTjuCvjHIInNzktHCgKQ5\nORAzI4JMPJ+GslWYHb4phowim57iaztXOoJwTdwJx4nLCgdNbOhdjsnvzqvHu7Ur\nTkXWStAmzOVyyghqpZXjFaH3pO3JLF+l+/+sKAIuvtd7u+Nxe5AW0wdeRlN8NwdC\njNPElpzVmbUq4JUagEiuTDkHzsxHpFKVK7q4+63SM1N95R1NbdWhscdCb+ZAJzVc\noyi3B43njTOQ5yOf+1CceWxG1bQVs5ZufpsMljq4Ui0/1lvh+wjChP4kqKOJ2qxq\n4RgqsahDYVvTH9w7jXbyLeiNdd8XM2w9U/t7y0Ff/9yi0GE44Za4rF2LN9d11TPA\nmRGunUHBcnWEvgJBQl9nJEiU0Zsnvgc/ubhPgXRR4Xq37Z0j4r7g1SgEEzwxA57d\nemyPxgcYxn/eR44/KJ4EBs+lVDR3veyJm+kXQ99b21/+jh5Xos1AnX5iItreGCc=\n-----END CERTIFICATE-----\n"
        }
    }
]
```

</details>

1. Import this flow into the Node-RED Editor
2. Confirm that the HTTP Request node has a TLS-Config that uses `ca109.pem`
3. Deploy
4. The request should succeed
5. Now deploy to an ESP32
6. The certificate isn't exported so the HTTP Request fails

ralphwetzel, 2023-08-07:
> 6\. The certificate isn't exported so the HTTP Request fails

I cannot confirm this:

I did a fresh install of the whole build system, imported the flow & build it for a simulator. The result - initially - is as expected:

* `flows.json` is created with all required nodes, including the `tls-config` node.
* `flows_cred_mcu.json` is created as well, holding the data of the certificate:

```
{
    "credentials": {
        "e35ea5b35fd6db16": {},
        "fbdc49382f0f04fa": {
            "cadata": "-----BEGIN CERTIFICATE-----\nMI[...]GCc=\n-----END CERTIFICATE-----\n"
        }
    }
}
```

* `flows.js` (created by `nodered2mcu`) yet doesn't get this data:

```
	nodes = flow.nodes();
	node = nodes.next().value;	// tls-config - fbdc49382f0f04fa
	node.onStart({
	});
```
* Consequentially the simulator raises with `Break: Resource: Resource not found: ca109.der!`

I'm suspecting the [check in line 1960](https://github.com/Moddable-OpenSource/moddable/blob/787bf7ccbf69aa1f9dc42cb44cf50ae33f2b1ac1/tools/nodered2mcu.js#L1957-L1972) of `nodered2mcu` to return `false` in case an embedded certificate is used - yet didn't analyze things deeper to have full evidence.

phoddie, 2023-08-07:
@ralphwetzel – thank you for the detailed analysis. It is good to see that you are getting the credentials exported correctly. I'm not sure why the certificate binary isn't being created. I'll reconfirm that works outside the plug-in first, to have some confidence than nodered2mcu is doing what I expect, then I can try to take a look using the plug-in. I'll let you know what I find.

phoddie, 2023-08-08:
Good news. It is entirely my bug. :(

I didn't anticipate the separate credentials file in the new code paths to extract data (like the TLS certificate) from the flow. I've fixed that. It will take a few days for that to propagate out as we are finishing up the August release.

Thanks for the help and apologies for the false alarm.

closed 2023-08-08 by phoddie