ralphwetzel/node-red-mcu-plugin #49: flow list mixed up between multiple projects

mshioji, 2024-03-24:
Hello,
This is only happen when "Projects feature" is turned on at Node-RED settings.js
[image]

how to reproduce:
1. if you have project-one with some flow-tabs like following, create new project-two from project menu. 
[image]
[image]

2. then we could see flow list of plugin is mixed up from project-one and project two.
however, it seems unnecessary flow options will not be shown after browser reload.
[image]

please note that flow-number count are slightly strange before following patch on node-red repo is applied:
https://github.com/node-red/node-red/issues/4600
https://github.com/node-red/node-red/pull/4619

Thank you,

ralphwetzel, 2024-04-06:
Hi @mshioji !
Thank you for this report. As I'm not using the project feature, I can't really test it.
I've yet tried to provide a fix - based on what @knolleary implemented to [reset the workspace index](https://github.com/node-red/node-red/pull/4619).
I'd appreciate of you could check if it solves the issue. You need to pull the latest changes directly from the repo to run this.

mshioji, 2024-04-08:
Hello @ralphwetzel ,
Thank you for the update. I also not using the project feature usually, but recently I learned this in yokoi-san's workshop and trying with node-red-mcu environment.

I tried plugin 1.5.0 today, now flow-tabs-list working fine. thank you for your work.
however, it seems build selection checkbox in the list still sometimes goes strange.
I am not sure following steps always reproduce the phenomenon but...
  1. Build Project two - flow 1
  2. switch to Project one
  3. touch checkbox on the flow-list
  -- "Deploy" button doesn't turn to RED, 
  -- Build button keep activated and I could start Build without flow selection.
   (this phenomenon recovered after reload the browser. )

[image]

Environment:
node v20.11.1
node-red v3.1.8
mcu plugin v1.5.0
OS: Ubuntu 22.04.4 LTS

Thank you,

ralphwetzel, 2024-04-12:
Hi @mshioji 
I think I found the reason for the "strange" behaviour you experienced.
The fix was straight forward and is already in the repo - awaiting your tests.

There's one open topic yet: Currently a running debug session will not be terminated when you switch projects. That might create the funny situation that you'll continue to see screen action when you're switching back to the project you started a build from.
Do you (and/or the people around you working with the plugin) have an opinion about what should happen with a running debug session when you switch projects? Keep it running - or terminate it?

Highly appreciate you feedback!

mshioji, 2024-04-13:
Hi @ralphwetzel ,
Thank you for your quick fix, I tried latest and it works very nice.

Regarding the debug session (Keep-running or terminate), my opinion is...  when I switch project, I think my brain already going to another project therefore terminate debug session is good in this case. 
I am asking this topic to some people so I think I will add comment if I get different answers.
Thank you very much!

phoddie, 2024-04-13:
> Do you (and/or the people around you working with the plugin) have an opinion about what should happen with a running debug session when you switch projects? Keep it running - or terminate it?

I'm not a user of Projects, but maybe the Node-RED behavior should guide here. If a flow is running in Node and you switch projects, what happens?

ralphwetzel, 2024-04-13:
Closing as resolved. Fix is implemented in [v1.5.1](https://github.com/ralphwetzel/node-red-mcu-plugin/releases/tag/v1.5.1).

closed 2024-04-13 by ralphwetzel

phoddie, 2024-04-14:
FWIW – the Node-RED [documentation](https://nodered.org/docs/user-guide/projects/) for the Projects features says:

> Node-RED only runs one project at any time. By opening another project you change what flows are running

This seems consistent with the comment of @mshioji above.