ralphwetzel/node-red-mcu-plugin #50: Dashboard no appearing in LCD

spillz-dxb, 2024-04-16:
Hi,

Being playing with this great piece of work, however ran into a small problem. While the flows work perfectly , except for some errors in xsbug, the dashboard dont appear on my m5stack fire and m5stack core2.

```
The environment is 
Welcome to Node-RED
===================

16 Apr 16:28:30 - [info] Node-RED version: v3.1.7
16 Apr 16:28:30 - [info] Node.js  version: v20.11.1
16 Apr 16:28:30 - [info] Windows_NT 10.0.22631 x64 LE
16 Apr 16:28:32 - [info] Loading palette nodes
16 Apr 16:28:34 - [info] Node-RED MCU Edition Runtime Version: #a02cd43
16 Apr 16:28:34 - [info] Node-RED MCU Edition Plugin  Version: v1.5.0
16 Apr 16:28:37 - [info] Moddable SDK Version: v4.6.0-0-g298fe72 (x86)
16 Apr 16:28:40 - [info] Dashboard version 3.6.5 started at /ui
16 Apr 16:28:41 - [info] Settings file  : C:\Users\sande\.node-red\settings.js
16 Apr 16:28:41 - [info] Context store  : 'default' [module=memory]
16 Apr 16:28:41 - [info] User directory : \Users\sande\.node-red
16 Apr 16:28:41 - [warn] Projects disabled : editorTheme.projects.enabled=false
16 Apr 16:28:41 - [info] Flows file     : \Users\sande\.node-red\flows.json

Error is `C:\Users\sande\Downloads\node-red-mcu-main\node-red-mcu-main\nodered.js (86) # Break: util.getSetting: cannot coerce undefined to object!`

Console output is 

`Starting build process...
Host system check: Windows 11 Home
MCU Build system check: p1.5.0 + #a02cd43 @ m4.6.0-0-g298fe72 (x86)
HOME directory check: C:\Users\sande
Creating build environment for platform esp32/m5stack.
Working directory: \Users\sande\.node-red\mcu-plugin-cache\6ml1pnfdqvl
$IDF_PATH is defined: C:\Users\sande\esp-idf-5.1.2\esp-idf
$IDF_TOOLS_PATH is defined: C:\Espressif
$LOCALAPPDATA is defined: C:\Users\sande\AppData\Local
> cd \Users\sande\.node-red\mcu-plugin-cache\6ml1pnfdqvl
UPLOAD_PORT = COM5
IDF_TOOLS_EXPORT_CMD = undefined
IDF_PYTHON_ENV_PATH = undefined
OPENOCD_SCRIPTS = undefined
ESP_IDF_VERSION = undefined
Creating build batch file...
> cmd.exe build.bat

C:\Users\sande\.node-red\mcu-plugin-cache\6ml1pnfdqvl>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat" 
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.43.0\cmd\
git version 2.43.0.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Users\sande\esp-idf-5.1.2\esp-idf

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.28.0.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.24.0\bin
[... 324 lines cut ...]
IDF_TOOLS_EXPORT_CMD = undefined
IDF_PYTHON_ENV_PATH = undefined
OPENOCD_SCRIPTS = undefined
ESP_IDF_VERSION = undefined
Creating build batch file...
cmd.exe build.bat
```

and

```
Not using an unsupported version of tool cmake found in PATH: 3.28.0.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
```

Also another point to highlight is 
```
NMAKE : warning U4004: too many rules for target 'C:\Users\sande\Projects\moddable\build\tmp\esp32\m5stack\debug\6ml1pnfdqvl\ftimage.h.xsi'
```
May not be an issue though.

With this resolution im hoping to continue my tinkering!

Thanks,
spillz

phoddie, 2024-04-16:
This is most likely what is preventing your flow from running:

> `C:\Users\sande\Downloads\node-red-mcu-main\node-red-mcu-main\nodered.js (86) # Break: util.getSetting: cannot coerce undefined to object!`

When that happens, xsbug should break so you can inspect the state. It would be helpful to see the JavaScript stack trace at that point.

Also, if you are able to share the JSON for a **simple** flow that reproduces this problem, it may help sort this out more quickly. Thank you.

spillz-dxb, 2024-04-16:
here is the flow

```
[
    {
        "id": "9c48a93517202288",
        "type": "tab",
        "label": "weather map",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": true
        }
    },
    {
        "id": "3403a3249a2ab867",
        "type": "openweathermap",
        "z": "9c48a93517202288",
        "name": "",
        "wtype": "current",
        "lon": "",
        "lat": "",
        "city": "",
        "country": "",
        "language": "en",
        "_mcu": {
            "mcu": true
        },
        "x": 130,
        "y": 260,
        "wires": [
            [
                "2aeef9be9b69b20f",
                "b6b5811bd41331b5",
                "d7eca73d6cf9f2e8"
            ]
        ]
    },
    {
        "id": "98669820fb07f988",
        "type": "ui_text",
        "z": "9c48a93517202288",
        "group": "f7e0493752f832d2",
        "order": 1,
        "width": "6",
        "height": "1",
        "name": "City",
        "label": "",
        "format": "{{msg.payload}}",
        "layout": "row-center",
        "className": "piu-text-city",
        "style": false,
        "font": "",
        "fontSize": "",
        "color": "#000000",
        "_mcu": {
            "mcu": true
        },
        "x": 530,
        "y": 60,
        "wires": []
    },
    {
        "id": "b6b5811bd41331b5",
        "type": "ui_text",
        "z": "9c48a93517202288",
        "group": "f7e0493752f832d2",
        "order": 4,
        "width": "6",
[... 160 lines cut ...]
        "tab": "0c757ed72ea6be1c",
        "order": 1,
        "disp": false,
        "width": "6",
        "collapse": false,
        "className": ""
    },
    {
        "id": "0c757ed72ea6be1c",
        "type": "ui_tab",
        "name": "Weather",
        "icon": "dashboard",
        "order": 1,
        "disabled": false,
        "hidden": false
    }
]
```

for this , aplologies im not a developer , you will have to guide me in this.

you share the JavaScript stack trace from that point.

Regards,
spillz

phoddie, 2024-04-16:
As a rule, third party nodes are not supported. In any case, please do not mix issues together. It creates confusion.

phoddie, 2024-04-16:
I just ran the weather example from the Node-RED MCU repository. It works just fine on ESP32. Here's what I did:

```
cd $NODEREDMCU/examples/dashboard/weather
mcconfig -d -m -p esp32/m5stack_fire ssid="my wifi" password="my wifi password"
```

In comparing the `$NODEREDMCU/examples/dashboard/weather/flows.json` and the flow you shared above (thank you!), there is an important difference: your flow does **not** contain a `ui_base` node. That node is required for all flows that use the dashboard to configure fonts, themes, etc. It is automatically created by the Node-RED editor when you use the dashboard.

If you are using the Node-RED MCU Plug-in to build and deploy your flow, the plug-in should automatically ensure that the `ui_base` node is present. If you are manually exporting the node from the Node-RED editor, you need to export **all** nodes, not just the node with your flow, to have the `ui_base` included.

ralphwetzel, 2024-04-17:
> Another point off topic , i used these nodes and doesnt work .
> https://github.com/bartbutenaers/node-red-contrib-blockly
> This will come in handy when we can reuse blockly codes.

Still off topic: Behind the scene, the `Blockly` node is "just" a fancy `function` node. Different UI, same runtime. I'll spend some time to check the effort of making it usable in `node-red-mcu`.

phoddie, 2024-04-17:
(To avoid going further off topic, I've opened a [Discussion for Blockly](https://github.com/phoddie/node-red-mcu/discussions/126) on the Node-RED MCU repository. Please continue this there.)

phoddie, 2024-04-18:
> noticed you were using linux

macOS, actually. But, the OS where Node-RED runs probably doesn't make a difference here.

From your notes, the steps you are taking to build, deploy, and run the Dashboard flows are not clear. Without that, neither @ralphwetzel nor I can reproduce the problem and so are limited in our ability to help resolve this issue. Would you share step-by-step instructions that we can follow to reproduce the issue? Thank you.

spillz-dxb, 2024-04-19:
Happy to provide what i did.

Hardware Devices : M5stack Core2 , M5stack Fire , M5Atom
This is just for POC as we need a bigger screen.
Node-RED version: v3.1.9
Node.js  version: v20.12.2
Node-RED MCU Edition Runtime Version: #a02cd43
Node-RED MCU Edition Plugin  Version: v1.5.1
Moddable SDK Version: v4.6.0-6-g3d6d0f8
Dashboard version 3.6.5

Flows: https://github.com/phoddie/node-red-mcu/blob/main/examples/dashboard/weather/flows.json

Followed the documentation at moddable for installing on esp32 devices and  individually tested , esp-idf ($idf_path\examples\get-started\hello_world) , moddable sdk ($moddable\examples\piu\balls) ,nod-red-mcu plugin through the node-red plugin and using mconfig in the plugin cache folder.
Tried in WIndows and then switched to Linux , Downgraded node.js to V14.7 and node-red-dashboard.
All compilations had no error , but the runtime threw this.

`/home/User/.node-red/node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/nodered.js (86) # Break: util.getSetting: cannot coerce undefined to object!`

Finally ended up looking into node-red-mcu and found out nodes/dashboard/ui_templates.js calling the fonts
` const FONT = config.DASHBOARD_FONT ?? RED.util.getSetting(node,'DASHBOARD_FONT') ?? "Roboto" ;`
Deleted the call and the runtime and it works . Right now im playing with the resolution and screen sizing.

Appreciate @phoddie for willing to help out , but it seems not to be a plugin error , would you recommend shifting this to node-red-mcu repo?.

phoddie, 2024-04-19:
Thanks for the additional info. I was eventually able to reproduce the problem. Simple fix. That's now committed. You just need to update/ireinstall the Node-RED MCU Plugin and you should be set.